SpacehuhnTech/esp8266_deauther #1263: ESP 01 Loading settings...Invalid Hash - reseted to default, then soft WDT reset

SinanAkkoyun, 2020-06-20:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

**Describe the error**
It seems that the non existing settings can not be accessed and the hash can not be verified.
The board is new, I even flashed the "Reset" script provided.

How can I fix it? Please help me, I have limited time till I need to use this, so any help is strongly appreciated.

```

Soft WDT reset

ctx: cont 
sp: 3fff1a00 end: 3fff1fb0 offset: 01b0

>>>stack>>>
3fff1bb0:  401038af 3ffee038 00000000 00000000  
3fff1bc0:  00002660 00000000 0000000a 00000000  
3fff1bd0:  00002660 00000000 0000000a 00000000  
3fff1be0:  0000000c 00000000 00000020 40104023  
3fff1bf0:  00000000 a0000000 00000000 0000001c  
3fff1c00:  00000000 00000004 02660000 0000a000  
3fff1c10:  00000000 00000000 00000000 40104f4e  
3fff1c20:  0000a000 000ac43d 00000000 3ffead8c  
3fff1c30:  4000050c 3fffc278 40001e65 3fff1c81  
3fff1c40:  3fff1d00 00002660 40104f68 40001e4b  
3fff1c50:  3fff1c80 00000000 00002660 00000000  
3fff1c60:  40205df6 00000030 00000004 ffffffff  
3fff1c70:  40205970 00000100 00000100 00000016  
3fff1c80:  39383234 00000100 3fff3450 00000005  
3fff1c90:  02000000 00000000 00000000 000eb100  
3fff1ca0:  3fff3728 3fff1fe0 00000000 00000030  
3fff1cb0:  40101a6d 000b12da 00000000 3ffead98  
3fff1cc0:  4000050c 3fffc278 40104f68 3fffc200  
3fff1cd0:  3fffd83c 40002394 00000004 00000000  
3fff1ce0:  40001da0 00000064 00000000 0000000a  
3fff1cf0:  00000000 00000003 3fff3660 00000001  
3fff1d00:  34323839 3fff3700 ffffffff 00000005  
3fff1d10:  02000000 3fff5cf3 04000102 3fff1fe0  
3fff1d20:  00000005 3fff1d50 00000001 00000000  
3fff1d30:  00000001 3fff4d2c 3fff1fe0 3fff3664  
3fff1d40:  00000000 3fff1db0 3fff1db0 3ffe9bd6  
3fff1d50:  40002514 3fffdd3c 00000000 00002650  
3fff1d60:  3fff1e30 00000003 00000008 00000000  
3fff1d70:  00000000 00002660 00000000 3fff1d00  
3fff1d80:  00000000 3fff1fe0 3fff3660 40205a33  
3fff1d90:  3fff1db0 3fff1db0 00000004 4010178e  
3fff1da0:  00002660 3fff3450 00000000 00000000  
3fff1db0:  3ffe9bd0 00002660 00000000 3fffbff0  
3fff1dc0:  00000000 00000000 3fff4d20 00000000  
3fff1dd0:  3fff3660 4010695e 3fff1ea0 00000001  
3fff1de0:  3fff1e30 00000001 3fff1fe0 4022aed6  
3fff1df0:  00002640 00000001 3fff1ea0 4022af2e  
3fff1e00:  3fff3660 00000001 3fff1ea0 4022b2f9  
3fff1e10:  3ffe8bc4 3fff1ea0 00000001 3fff0c20  
3fff1e20:  3ffe8bc4 3fff1ea0 00000001 4022b378  
3fff1e30:  3ffe00ff 3fff1ea0 3fff1e50 40228ce5  
3fff1e40:  3fff4cc0 3fff1ea0 00000001 4021d03a  
3fff1e50:  3ffe97d0 00000000 000003e8 4022af5e  
3fff1e60:  3fff4d20 3fff4d70 3fff1e90 4022b060  
3fff1e70:  3ffe8bc4 00000000 3fff1f00 4022b084  
3fff1e80:  3fff4cb0 3fff1ef0 3fff1f00 4021e405  
3fff1e90:  3fff4d00 0000000f 0000000b 4022aed6  
3fff1ea0:  00000000 0000263f 0000263f 4022af2e  
3fff1eb0:  00000000 3fff1f00 3fff1f00 4022af5e  
3fff1ec0:  00000000 3ffe8a38 3fff1f00 4022b060  
3fff1ed0:  00000000 3ffe8a38 3fff0b50 3fff0c20  
3fff1ee0:  00000000 3ffe8a38 3fff0b50 402159c3  
3fff1ef0:  3ffe8a38 0000001c 00000000 00000fa0  
3fff1f00:  3fff4cc0 0000000f 0000000b 3fff4cc0  
3fff1f10:  0000000f 0000000b 3fff4d00 0000000f  
3fff1f20:  00000002 3fff4d20 0000000f 00000001  
3fff1f30:  3fff4ce0 0000000f 00000001 000000ff  
3fff1f40:  4023530c 3fff0b64 3fff0eb8 00000000  
3fff1f50:  00000001 4023533a 00000001 3fff0c20  
3fff1f60:  00000000 3fff0eb8 3fff0b64 4021e6fe  
3fff1f70:  499602d2 00000002 499602d2 00000001  
3fff1f80:  00000000 00000000 00000000 3fff0f7c  
3fff1f90:  3fffdc20 00000000 3fff0f75 402250d2  
3fff1fa0:  00000000 00000000 3fff0f90 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Mounting SPIFFS...OK
Loading settings...Invalid Hash - reseted to default

```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 1.0]
 - Hardware [e.g. DSTIKE, DIY]

**Attempts**
What have you already tried and didn't work out.

**Additional context**
Provide as much information as possible, better too much than too little!
If you don't use this template, your issue might be closed and tagged invalid!

spacehuhn, 2020-06-20:
First of all, hello and thanks for using the issue template! \o/

One thing you could quickly try out is by installing the [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder) then flashing it with Arduino and pasting the serial output (when it crashes) into the EspExceptionDecoder.

PeterZhang0820, 2020-07-10:
# The same problem is bothering me.  #
**My developing environment:**

Mac OS 10.14.4 with Arduino 1.8.12 

**My Board Model:**

NodeMCU 1.0  with CP2102 (ESP-12E)

**Tools I have:**

ESPlorer  ESP8266FLASH  PuTTY

**What I've tried:**

I have already search on the issues already closed, My problem is simular to [#1266] (https://github.com/SpacehuhnTech/esp8266_deauther/issues/1266), so I tried,
>The problem was that the board ESP8266Boards (2.7.1) -- >NodeMCU 1.0 was not chosen correctly.
RIGHT ESP8266 Deauther Modules -- > NodeMCU 1.0.
 
 But problem still exist. So I went futher more, I found this one [#1030] (https://github.com/SpacehuhnTech/esp8266_deauther/issues/1030)   

 >I got it working now. I used 4MB file when flashing the firmware, while flash size under tools was only set to 1MB... @ricolay you may want to check if you have same problem.
 
Although they recomanded 1M(256KB SPIFFS), I changed many cobination such as 4M(3MB SPIFFS)
I also changed borad type. By It didn't work. And that issue became inactive, so I did't get what I want.

I just change the screen from 0.96 to 1.13, So I undo everything but it was no help.

**Describe the error**

The board is new, When I upload the [Beacon_Spam] (https://github.com/spacehuhn/esp8266_beaconSpam), It work perfectly well, But at this project, after flashing .BIN file problem still exist.

    Can't autodetect firmware, because proper answer not received (may be unknown firmware). 
    Please, reset module or continue.
    K
    Formatting EEPROM...OK
    Loading settings...Invalid Hash - reseted to default
    Device names loaded from /names.json
    SSIDs loaded from /ssids.json
    Scan results saved in /scan.json
    Serial interface enabled
    Started AP
    [WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false,  captive-portal: true
    STARTED! \o/
    2.2.0
     # scan -t 5s
     Stopped scan
    Scan results saved in /scan.json
    Removed all APs
    Cleared station list
    Starting scan for access points (Wi-Fi networks)...

     Exception (3):
    epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4025a3fb depc=0x00000000

     ctx: cont 
    sp: 3fff1b10 end: 3fff1e20 offset: 01a0

    >>>stack>>>
    3fff1cb0:  3ffe87a8 4025a3fb 3fff1dd0 4022b44c  
    3fff1cc0:  4025a3fb 4010695e 3fff1dd0 3ffefcd8  
    3fff1cd0:  0000565f 3fff09dc 3fff1dd0 402186ae  
    3fff1ce0:  00000000 00000000 00000000 4022b002  
    3fff1cf0:  0000565f 3fff09dc 3fff1dd0 402189b5  
    3fff1d00:  0000565f 3fff09dc 3fff0a98 40218dcd  
    3fff1d10:  00000202 00000001 0000003c 00000000  
    3fff1d20:  00000258 00000119 00000001 00000001  
    3fff1d30:  fc100001 00565eca 0ae0ce24 00c80003  
    3fff1d40:  77700003 0064656e 00000000 00000000  
    3fff1d50:  00000000 00000000 00000000 00000000  
    3fff1d60:  64000000 74756165 00726568 00000000  
    3fff1d70:  00000000 00000000 00000000 00000000  
    3fff1d80:  00000000 00000000 00000000 00000000  
    3fff1d90:  00000000 00000000 00000000 00000000  
    3fff1da0:  00000000 04a8c000 00010101 01006e65  
    3fff1db0:  00000101 00000000 00000258 3802be89  
    3fff1dc0:  ec010af3 a76e19cf b169c64a 61677a26  
    3fff1dd0:  3fff6a10 0000025f 00000002 3ffefcd8  
    3fff1de0:  0000565f 3ffefcd4 3fff09dc 4021c878  
    3fff1df0:  00000000 00000000 00000000 3fff0df4  
    3fff1e00:  3fffdc20 00000000 3fff0ded 40225189  
    3fff1e10:  00000000 00000000 3fff0e00 40100114  
    <<<stack<<<

    ets Jan  8 2013,rst cause:2, boot mode:(3,6)

    load 0x4010f000, len 1264, room 16 
    tail 0
    chksum 0x42
    csum 0x42
    ~ld

    Mounting SPIFFS...OK
    Loading settings...OK
    Device names loaded from /names.json
    SSIDs loaded from /ssids.json
    Scan results saved in /scan.json
    Serial interface enabled
    Started AP
    [WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false,  captive-portal: true
    STARTED! \o/
    2.2.0
     # scan -t 5s
    Stopped scan
    Scan results saved in /scan.json
    Removed all APs
    Cleared station list
    Starting scan for access points (Wi-Fi networks)...

     Exception (3):
     epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4025a3fb depc=0x00000000

    ctx: cont 
    sp: 3fff1b10 end: 3fff1e20 offset: 01a0

    stack
    3fff1cb0:  3ffe87a8 4025a3fb 3fff1dd0 4022b44c  
    3fff1cc0:  4025a3fb 4010695e 3fff1dd0 3ffefcd8  
    3fff1cd0:  0000029f 3fff09dc 3fff1dd0 402186ae  
    3fff1ce0:  00000000 00000000 00000000 4022b002  
    3fff1cf0:  0000029f 3fff09dc 3fff1dd0 402189b5  
    3fff1d00:  0000029f 3fff09dc 3fff0a98 40218dcd  
    3fff1d10:  00000202 00000001 0000003c 00000000  
    3fff1d20:  00000258 00000119 00000001 00000001  
    3fff1d30:  fc100001 00565eca 0ae0ce24 00c80003  
    3fff1d40:  77700003 0064656e 00000000 00000000  
    3fff1d50:  00000000 00000000 00000000 00000000  
    3fff1d60:  64000000 74756165 00726568 00000000  
    3fff1d70:  00000000 00000000 00000000 00000000  
    3fff1d80:  00000000 00000000 00000000 00000000  
    3fff1d90:  00000000 00000000 00000000 00000000  
    3fff1da0:  00000000 04a8c000 00010101 01006e65  
    3fff1db0:  00000101 00000000 00000258 3802be89  
    3fff1dc0:  ec010af3 a76e19cf b169c64a 61677a26  
    3fff1dd0:  3fff6988 0000025f 00000002 3ffefcd8  
    3fff1de0:  0000029f 3ffefcd4 3fff09dc 4021c878  
    3fff1df0:  00000000 00000000 00000000 3fff0df4  
    3fff1e00:  3fffdc20 00000000 3fff0ded 40225189  
    3fff1e10:  00000000 00000000 3fff0e00 40100114  
    <<<stack<<<

    ets Jan  8 2013,rst cause:2, boot mode:(3,6)

    load 0x4010f000, len 1264, room 16 
    tail 0
    chksum 0x42
    csum 0x42
     ~ld
These code just shows again and again and onboard LED blinks almost once per second.

Sorry for the ugly typography and grammatical errors.

SinanAkkoyun, 2020-07-19:
Hey, thank you for your awnser! I really like your work.

I will try it in some days and send you the log.

andynemesis3, 2020-08-15:
Getting same error, 

load 0x4010f000, len 3664, room 16 
tail 0
chksum 0xee
csum 0xee
v00000000
~ld

Mounting SPIFFS...OK
Loading settings...OK
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: true
STARTED! \o/
2.2.0
# scan -t 5s
Stopped scan
Scan results saved in /scan.json
Removed all APs
Cleared station list
Starting scan for access points (Wi-Fi networks)...

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x402702d7 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffcd0 end: 3fffffc0 offset: 0190
3ffffe60:  3ffe0022 3ffffe80 402702d7 402258af  
3ffffe70:  3ffe8888 402702d7 3fffff70 40212840  
3ffffe80:  2e322e00 006e0030 80000000 402258af  
3ffffe90:  3fff11e1 3ffeec64 3fffff70 40212ad8  
3ffffea0:  3fff11e1 3ffeec64 3ffeed18 40212f29  
3ffffeb0:  00000202 00000001 0000003c 00000000  
3ffffec0:  00000258 00000119 00000001 00000001  
3ffffed0:  2d160001 00f24f62 8eba2716 00c8005b  
3ffffee0:  77700003 0064656e 00000000 00000000  
3ffffef0:  00000000 00000000 00000000 00000000  
3fffff00:  64000000 74756165 00726568 00000000  
3fffff10:  00000000 00000000 00000000 00000000  
3fffff20:  00000000 00000000 00000000 00000000  
3fffff30:  00000000 00000000 00000000 00000000  
3fffff40:  00000000 04a8c000 00010101 01006e65  
3fffff50:  00000101 00000000 00000258 d1c1c4bb  
3fffff60:  3b77fdd5 ccb26283 2379a4a9 4dec1a59  
3fffff70:  3fff353c 0014025f 00fef024 3ffee740  
3fffff80:  00000163 3ffee73c 3ffeec64 4021a740  
3fffff90:  feefeffe feefeffe feefeffe 3ffef064  
3fffffa0:  3fffdad0 00000000 3ffef024 40226b04  
3fffffb0:  feefeffe feefeffe 3ffe853c 40101111  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

### **Using exception Decoder:**

Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x4000bf64
EXCVADDR: 0x402702d7

Decoding stack results
0x402258af: String::String(char const*) at C:\Users\andy\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.7.2\cores\esp8266\WString.cpp line 34
0x40212840: jsonFlag(String&, char const*, bool) at C:\Users\andy\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.7.2\cores\esp8266/WString.h line 124
0x402258af: String::String(char const*) at C:\Users\andy\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.7.2\cores\esp8266\WString.cpp line 34
0x40212ad8: Settings::getJsonStr() at C:\Users\andy\AppData\Local\Temp\arduino_build_219860\sketch\Settings.cpp line 101
0x40212f29: Settings::save(bool) at C:\Users\andy\AppData\Local\Temp\arduino_build_219860\sketch\Settings.cpp line 193
0x4021a740: loop() at D:\Projects\Arduino\esp8266_deauther\esp8266_deauther/esp8266_deauther.ino line 184
0x40226b04: loop_wrapper() at C:\Users\andy\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.7.2\cores\esp8266\core_esp8266_main.cpp line 197

shojkeee, 2020-12-19:
The problem is in the board manager. you probably installed deauther esp8266 boards 2.7. x ver. It will work with Deauther ESP8266 Boards ver.2.0.0