SpacehuhnTech/esp8266_deauther #1271: What do "RST" and "FLASH" buttons do and how do you use them?

turkeyphant, 2020-07-08:
I have the v3.5 board. There are two buttons that I can't find any documentation for.

(Also, I cannot get the device to be recognised over USB in order to flash the new software, is it something to do with this?)

turkeyphant, 2020-07-08:
I have followed these steps.

vuongfcc, 2020-07-11:
> I have the v3.5 board. There are two buttons that I can't find any documentation for.
> 
> (Also, I cannot get the device to be recognised over USB in order to flash the new software, is it something to do with this?)

https://www.hackster.io/234v244/esp8266-wifi-deauthentication-attack-4ed3d3

turkeyphant, 2020-07-11:
> https://www.hackster.io/234v244/esp8266-wifi-deauthentication-attack-4ed3d3

What are you referring to in that post? There's no mention of RST and the only mention of FLASH is for different firmware.

huangguiniab, 2020-08-15:
1，RST按钮连接RST信号，当这个按钮按下时，RST脚电平被拉低，开发板就会重启，这就是这个按钮的用处
2，FLASH按钮连接GPIO0，平时GPIO0都是悬空状态，通电时开发板正常工作，当按下这个按钮通电时(注意：是通电前按下)，开发板就会进入烧录模式，此时就可以正常烧录程序进入开发板。我想，你肯定没有按下那个按钮直接烧录了，这就是你烧录失败的原因。
3，对不起我不会说讲英语，让你阅读很困难，希望我的回答能帮到你😀

turkeyphant, 2020-08-15:
I haven't tried to flash at all.

When you say the RST makes the board restart how long do you have to hold it for? My RST button seems to do nothing at all.

huangguiniab, 2020-08-17:
我的开发板是ESP-12E，按一下USB口旁边的那个RST按钮它就重启了，不需要保持也不需要要按住很久。可能我的开发板和你的按键定义不一样吧
[image]

turkeyphant, 2020-08-17:
Thanks. Nothing happens at all when I press the RST button on my board :(

(see also https://github.com/SpacehuhnTech/esp8266_deauther/issues/1272)

huangguiniab, 2020-08-17:
我看了 #1272 了，我从你的描述中知道了您的开发板是dstike 的开发板，但是我在他们的网站上个淘宝网上都找不到和你照片中开发板一模一样的型号，但找到了差不多的OLEDv6的pdf手册https://github.com/lspoplove/Deauther-Project/blob/master/Deauther-OLED/DeautherOLEDV6/DeautherOLEDV6.pdf，我猜你这个可能和OLEDv6的电气设计一样。首先那四个灯的问题，从pdf上看，它们是相互独立的
[image]并且只连接到“IP5306这个芯片上，我怀疑是这个LED灯珠坏了或者是这个芯片坏了才导致有一个LED灯不亮，你用万用表或者是能测电压的仪表看看灯珠两头有没有3v左右的电压。其次就是你这个RST键的设计问题
[image]
[image]你这个板子的设计和我的一样，nRST信号被按钮接入负极，按下按钮确实是重启，没重启可能是按钮坏了，你拿个东西短路按钮两边试试，屏幕不显示的话测上图框出的esp的8-9脚和下图屏幕的8-9脚有没有3-4v的电压
[image]
现在是我这的凌晨2点钟了，我得赶紧睡觉，不能让我老爸发现我玩手机😁

turkeyphant, 2020-08-18:
Thanks for more help. I think you're right about the broken green battery LED.

But do you know what's causing the red LED? I think this is related to why it won't turn on any more.

huangguiniab, 2020-08-18:
不，只要那个芯片通了电那个红色的LED就会亮，启动不了与那个灯无关。
我现在突然想起来，这个固件的设置里可以禁用掉OLED屏幕，也许你的开发板并不是没有启动，而是OLED屏幕被禁用掉了，你用手机搜索一下WI-FI信号，有没有一个叫pwned的热点，也许可以通过web控制台开启OLED屏幕。

turkeyphant, 2020-08-18:
For a second there I thought you must have the answer! Unfortunately it didn't work and I couldn't see the network pwned.

It also powers off automatically after 30 seconds. This is not normal behaviour.

And the RGB LED does not work at all :(

huangguiniab, 2020-08-18:
1. 从你的描述里看，那四个灯亮了30秒就灭掉，是因为IP5306实质上就是一个充电宝芯片，它检测后面通过的电流太少了所以就自动切断了电源，开发板无法启动与这个无关。

2.我仔细想了想，现在想修好你这个开发板比较困难，因为你没有能测量电压，测量电阻的仪表，无法检查这块开发板上面的信号,电压是否正常。而且你还需要一把电烙铁和一些焊锡才能修好它，这非常考验手工能力，比较复杂，我建议您直接放弃这个开发板，买一个ESP-12E，这个才卖$2，很便宜，但如果您真的要修好它，你需要买很多东西

3.我可以解答wifi芯片上的两个灯的用途
红色LED:电源指示灯，一旦开发板通电它就会亮
蓝色LED:由程序控制的灯泡，它不亮是正常的

4.至于RGB LED的问题，它是由程序控制的，开发板没有正常启动，它自然不会亮

lspoplove, 2020-08-18:
为Naver Mail应用程序发来的。

副本: "Subscribed" <subscribed@noreply.github.com>
Sent: 2020.08.19 上午 00:48:27 (GMT+08:00)

closed 2020-11-24 by stale[bot]