SpacehuhnTech/esp8266_deauther #1272: Won't turn on (red LED and three battery LEDs)

turkeyphant, 2020-07-09:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

**Describe the error**
Charged via USB.
During charging third battery LED (from the top) went off.
Now won't turn on.

One power press makes the three battery LEDs shine green:

ON
ON
OFF
ON

And a red LED above the chip.

The main RGB LED doesn't even flicker and the display is blank.

It turns off after about 15 seconds.

Turning off with double press of power switch also works.

**Expected behavior**
Works normally. Turns on and off. Menu and features work,

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Hardware DSTIKE v3.5

lspoplove, 2020-07-09:
> 副本: "Subscribed" <subscribed@noreply.github.com>
> 主旨: [SpacehuhnTech/esp8266_deauther] Won't turn on (red LED and three battery LEDs) (#1272)

turkeyphant, 2020-07-11:
Still stuck in this mode and won't turn on.

turkeyphant, 2020-08-15:
[image]

Bommlon, 2020-08-31:
Have you tried re-flashing the board? If this doesn't help it could be a problem with the board. Try contacting DSTIKE if that's the case.

turkeyphant, 2020-09-03:
I have tried reflashing the board but neither of the two recommended software work (have tried on a Windows 10 PC and a macOS laptop).

esptool can't connect to it in both instances for some reason.

closed 2020-12-09 by stale[bot]