SpacehuhnTech/esp8266_deauther #1273: unable to compile and upload using Arduino IDE

helmarw, 2020-07-13:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes
I followed the instruction here using a clean install of Arduino IDE 1.8.13
https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide
thought the board selection look different form the recent board files, so i choose Generic 8266
this are the settings i used:
[image]

for testing i used the release source code 2.1.0:
https://github.com/SpacehuhnTech/esp8266_deauther/archive/v2.1.0.zip
and the config for DSTIKE_OLED_v3.5_Monster

im using a Deauther OLED v5 with preinstalled 2.1.0 and i also can successfully load the release bin with esptools

**Describe the error**
But im not able to flash the files using the Arduino IDE
What do you think causes it? I have no idea

```
Error/Compile/Output Log:  no compiler error
but hangs in a boot loop, an yes i tried erasing flash too and also used the reset_4MB.bin

Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 9
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '9', hidden: false, captive-portal: true
STARTED! \o/
v2.1.0

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4026f883 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd90 end: 3fffffc0 offset: 0190
3fffff20:  00000000 4bc6a7f0 4026f883 40226c4b  
3fffff30:  40220124 3ffee7a4 3ffee790 4020af68  
3fffff40:  00000000 3ffee614 40100766 00418900  
3fffff50:  00000000 80000000 3ffef6cc 3ffeeef8  
3fffff60:  3fffdad0 3ffee614 3ffee790 4020d607  
3fffff70:  3fffdad0 3ffee614 3ffee790 4020d662  
3fffff80:  3fffdad0 3ffee614 3ffeeeb8 4021ba5d  
3fffff90:  feefeffe feefeffe feefeffe 3ffeeef8  
3fffffa0:  3fffdad0 00000000 3ffeeeb8 40227ea0  
3fffffb0:  feefeffe feefeffe 3ffe854c 401011e9  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3664, room 16 
tail 0
chksum 0xee
csum 0xee
v00000000
~ld

```

**To Reproduce**
always the same when flashing with Ardunio IDE and above settings

**Environment (please complete the following information):**
 - OS: MacOS Catalina
 - Browser Safari
 - Version [e.g. 1.0]
 - Hardware DSTIKE OLED V5

**Attempts**
reading, trying to reset, erase flash using rest.bin, changing some settings (frequency, baud rate, number of LEDs (dont know which is the right number anyways), flashing sketch only and also all content) a.s.o.
i also tried first compiling the bin files and then flashing it with esptools
nothing worked so far

any suggestions are welcome
thnx !

helmarw, 2020-07-14:
update:
esptools sais it detects 4MB Flash but according to your wiki its an ESP-07 so it has only 1MB
i tried compiling with 1M (256K SPIFFS) but it gives me an compiler error

```
/Users/helmar/Library/Arduino15/packages/deauther/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: esp8266_deauther.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'
collect2: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266.
```

the only possible option is 1M (192K SPIFFS) but again when doing so it ends up in a boot loop

additionally i tried filesystem uploader (https://github.com/esp8266/arduino-esp8266fs-plugin) to upload the /data directory before and after flashing the sketch. 

Still no luck ....

helmarw, 2020-07-14:
update2:

i tried to find out at which point the code crashes
so i added a few println in the setup and main loop

`    // setup LED
    led.setup();
    prntln("setup LED done");
}

void loop() {
    currentTime = millis();
    prntln("time update done");
    led.update();    // update LED color
    prntln("LED update done");
    wifiUpdate();    // manage access point
    prntln("WIFI update done");
    attack.update(); // run attacks
    prntln("ATTACK update done");
    displayUI.update();
    prntln("displayUI update done");
    cli.update();    // read and run serial input
    prntln("CLI update done");
    scan.update();   // run scan
    prntln("SCAN update done");
    ssids.update();  // run random mode, if enabled
    prntln("SSID update done");`

not i see in the serial output that it crashes after finishig attack.update or envoking displayUI.update();

`Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 9
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '9', hidden: false, captive-portal: true
STARTED! \o/
v2.1.0
setup LED done
time update done
LED update done
WIFI update done
ATTACK update done

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4026f8e3 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd90 end: 3fffffc0 offset: 0190
3fffff20:  00000000 4bc6a7f0 4026f8e3 40226ca7  
3fffff30:  40220180 3ffee844 3ffee830 4020af68  
3fffff40:  00000000 3ffeee4c 40100766 68312600  
3fffff50:  00000000 80000000 3ffeff54 3ffeef98  
3fffff60:  3fffdad0 3ffee6b4 3ffee830 4020d607  
3fffff70:  3fffdad0 3ffee6b4 3ffee830 4020d662  
3fffff80:  3fffdad0 3ffee6b4 3ffeef58 4021baa1  
3fffff90:  feefeffe feefeffe feefeffe 3ffeef98  
3fffffa0:  3fffdad0 00000000 3ffeef58 40227efc  
3fffffb0:  feefeffe feefeffe 3ffe854c 401011e9  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3664, room 16 
tail 0
chksum 0xee
csum 0xee
v00000000
~ld`

any idea why ?

helmarw, 2020-07-14:
update3:

i only disabled the line 
 //   displayUI.update();

now its working, less of course the display. dont know what wrong there, must be wrong settings in the A_Config ?!
but why is the corresponding deauther_2.1.0_DSTIKE_OLED_v3.5_v4_Monster.bin working
so my guess is settings in A_config of derectroy DSTIKE_OLED_v3.5_Monster are wrong ?!

```
#ifndef config_h
#define config_h

// ===== DISPLAY LIBRARY ===== //
#include "SH1106Wire.h"
// =========================== //

// ===================== LED CONFIG ==================== //
#define NEOPIXEL_LED

#define LED_MODE_BRIGHTNESS 10 // brightness of LED modes 

#define LED_NEOPIXEL_NUM 1
#define LED_NEOPIXEL_PIN 15
#define LED_NEOPIXEL_MODE NEO_GRB + NEO_KHZ800

// ===================== DISPLAY CONFIG ==================== //
#define USE_DISPLAY true // default display setting

#define BUTTON_UP 12
#define BUTTON_DOWN 13
#define BUTTON_A 14

#define DEAUTHER_DISPLAY SH1106Wire display = SH1106Wire(0x3c, 5, 4); // for 1.3" OLED
```

does anybody know what the problem here is ?

XXCoreRangerX, 2020-07-18:
I got the same problem. Even when enabling the display from the WEB Interface it gets into a bootloop.
> thought the board selection look different form the recent board files
It's because the libraries have been **updated**. The previous library was **arduino-esp8266-deauther by Tobozo**, which worked **perfectly fine**. I don't know why they replaced it with Spacehuhn's libraries, but since then pretty much everything is failing.

XXCoreRangerX, 2020-07-18:
I think they should bring back the older libraries, had **no issues** with them whatsoever.

helmarw, 2020-07-19:
thnx for the info, do you have a link to the old libraries ?
somebody must compiling the release bins, and they are working, so what libraries are they using ?
I would have thought they are using the latest possible to see that everything is working ....

XXCoreRangerX, 2020-07-19:
You can still download older libraries source code from the GitHub page: https://github.com/tobozo/Arduino
I don’t really know if it would work, I don’t have a lot of experience with that. We should wait for @spacehuhn to reply.

spacehuhn, 2020-07-19:
I am working on some bug fixes, but I have no issues with the OLED libraries:
https://github.com/SpacehuhnTech/esp8266_deauther/releases/tag/2.5.0-rc1

helmarw, 2020-07-19:
yes looks like it since your release bins are working, but even after a clean install it does not work for me, i tried 3 different boards/assemblies with OLEDs
did you try also a clean install just following the instructions in the wiki ?
maybe there are steps missing ?!

XXCoreRangerX, 2020-07-20:
The **easier** way to fix this problem would be to just bring @tobozo libraries back, they worked **perfectly fine, as I stated above a few times,** and there was no need to update them. I have the old library files but I don't know how to set up package_deauther_index.json properly with older configuration.
The **better** way to fix this problem would be fixing newer packages.

tobozo, 2020-07-20:
@XXCoreRangerX I've notified @spacehuhn about an idea where both custom SDK versions would be available as a choice in the boards manager.

I can share this setup but it's totally untested on windows and mac.

[image]

XXCoreRangerX, 2020-07-20:
That’s a really good idea

tobozo, 2020-07-20:
knock yourself out on this *temporary test file* => https://phpsecu.re/esp8266/package_deauther_all_index.json

spacehuhn, 2020-07-20:
[image]

I wasn't planning on using the new Arduino package for the 2.0, 2.1 or 2.2 versions.
But like I said, I **am** working on [version 2.5](https://github.com/spacehuhntech/esp8266_deauther/tree/v2.5) to includes some bug fixes and there are more are to come.
There was supposed to be a slow transitioning. The new Arduino core is not fully done yet - you can use it already, it works but I will make more changes to add some libs and boards soon.

And yes I will add the 2.0.0 to the packet manager in the next few days.

satan666000, 2020-09-23:
In Arduino the same error when flashing and it does not turn on, so I recommend that you use your android phone and that is easier

closed 2020-12-29 by stale[bot]