SpacehuhnTech/esp8266_deauther #1274: Not able to get any output from 8266 when connected to iPhone

jwothke, 2020-07-13:
I have burned an 8266 on my Windows 10 machine and used to with no issues.  I then connected it to my Samsung Note and it worked fine on that device.  When I connect to my iPhone I get nothing.  I tried 4 different cables from 3 different vendors and nothing.  I buzzed the pins on the dev board and nothing on 3.3 or Vin etc. Also tried multiple 8266 and even re-burned one to see if that may wake it up...nothing.
I have searched high and low but nothing.

I think that I must be overlooking a setting on the iPhone?  Not real sure......

If I could get a push in a direction that would be great.

Thanks
```

Bommlon, 2020-07-14:
Sounds like a Problem with the iPhone. May be a security measure by Apple, which would be hard to circumvent.

jwothke, 2020-07-15:
Thanks Bommlon.

I was thinking it might be something with the fone.  I just thought I would see something while searching where someone else might have run across it.

If I dont hear much back here, I will take a look on the Apple boards. (yikes!!!)

Bommlon, 2020-07-15:
As far as I know the deauther would only use the phone for supplying power. Maybe try a power-only cable.

helmarw, 2020-07-16:
you cant use serial over cable on the iphone unless you jailbreak but it may supply power so you can connect to the AP

jwothke, 2020-07-16:
helmarw,

makes sense what you mentioned about needing to jailbreak.  If i do NOT JB, I woud think I would NOT be able to get the AP (web interface at 192.168.4.1) to work correct?  That would be all signal back to iPhone correct?

thanks again

helmarw, 2020-07-19:
no AP is working, just tested it, no JB needed (why would I, its just wifi) but no need for a cable connection unless you want to use your iphone as power supply. that doesnt work with all models, so better use battery
The only reason why you would want to connect with cable ist using the serial interface, but that may only works on JB devices, if it works...

jwothke, 2020-07-24:
OK!!!!  helmarw........you are 100% correct.  I hooked up the 8266 to a power supply and sure enough nothing else is needed.  My guess is that the cables that I purchased didnt have power to Vin.  I tried 2 of the cables and measured no 3.3 on the 8266.  So even though I purchased from multiple vendors, the cables are made with no power to the 8266 from the Apple iPhone.

Thank to everyone.

closed 2020-07-24 by jwothke