SpacehuhnTech/esp8266_deauther #1276: Display

ryuk171, 2020-07-16:
Esp8266 nodemcu amica display i2c ssd1306 not working
 V2.2 is not flashing properly w/ defective web gui

XXCoreRangerX, 2020-07-18:
I've been getting similar errors since the library update. The previous arduino-esp8266-deauther package by Tobozo was working fine.

ryuk171, 2020-07-29:
Yeah i tried the old one with v2.0 board manager ,works fine w/ v2 of deauther but not w/ v2.5

joopy, 2020-10-24:
Yes the same here ssd1306 not working in v2.5

Is this correct?
`#if !defined(SSD1306_I2C) && !defined(SSD1306_SPI) && !defined(SH1106_I2C) && !defined(SH1106_SPI)
  #define SSD1306_I2C
  #define USE_DISPLAY false
#else /* if !defined(SSD1306_I2C) && !defined(SSD1306_SPI) && !defined(SH1106_I2C) && !defined(SH1106_SPI) */
  #define USE_DISPLAY true
#endif /* if !defined(SSD1306_I2C) && !defined(SSD1306_SPI) && !defined(SH1106_I2C) && !defined(SH1106_SPI) */`

closed 2021-01-29 by stale[bot]

joopy, 2021-01-30:
still open?

1jammer1, 2021-02-01:
ok?

-- 

_____________________________________________________________

 
CONFIDENTIALITY/PRIVACY NOTICE: This message and any attachments 
transmitted
with it is for the designated recipient only and may contain 
privileged or
confidential information. If you have received it in error 
please notify the
sender, via return e-mail, immediately and permanently 
delete the original. Any
unauthorized review, disclosure, dissemination, 
distribution, or copying of this
e-mail is strictly prohibited. Thank you.