SpacehuhnTech/esp8266_deauther #1277: Scanning Station Error

slashblood, 2020-07-18:
Hey guys, the deauther is working fine on me but i just have a bit of a problem, when i click the `scan station` it displays an error saying
`ERROR: loading file: run?cmd=scan stations -t 15s -ch all`
How to fix this error?

avlhack1, 2020-07-24:
Hola, yo tengo el mismo problema, al presionar stations se detiene todo y me sale error loading ClientScan.json?time=15 
Cambio el valor de scan time pero me da el mismo error.

spacehuhn, 2020-07-24:
The access point has to shut down for a scan and will become available again after the scan is done
https://github.com/SpacehuhnTech/esp8266_deauther/wiki/FAQ#timeout-error

Bommlon, 2020-07-25:
> Hola, yo tengo el mismo problema, al presionar stations se detiene todo y me sale error loading ClientScan.json?time=15
> Cambio el valor de scan time pero me da el mismo error.

Hi! Sadly our Spanish-skills range from barely to nonexistent. Could you please reply in English instead?

closed 2020-10-30 by stale[bot]