SpacehuhnTech/esp8266_deauther #1279: How to factory reset deauther watch to default?

manishrd, 2020-07-23:
Just now received Deauther watch from Aliexpress. I flashed it via ESP8266 flasher but after flashing the navigation wheel messed up. Now it selects upon scrolling the wheel down. It scrolls by pressing the wheel. another issue the LED went off forever after flashing successfully. 

Please help me get the watch back to default or factory setting ho it came. I dont want to keep the newly flashed thing.

Please help.

spacehuhn, 2020-07-23:
@manishrd try flashing the bin file, for example this one: https://github.com/SpacehuhnTech/esp8266_deauther/releases/download/2.5.0-rc1/esp8266_deauther_2.5.0_rc1_DSTIKE_DEAUTHER_WATCH.bin

mrvodka007, 2020-07-24:
You can also completely wipe the memory, which I find works more often by running esptool with these arguments: `--port [Port] --baud 115200 erase_flash`

manishrd, 2020-07-24:
Thank you guys.. its done now. Much appreciated!

closed 2020-07-24 by manishrd

unbannedez, 2022-01-07:
Hello how do I know wich port it is?