SpacehuhnTech/esp8266_deauther #1283: Unable to connect with OpenWrt (in client mode)

eisengrau, 2020-07-28:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

**Describe the error**
A clear and concise description of what the problem is.
What do you think causes it?

Hello,

I would like to set this up as a permanent member of my network, so I can access the web interface from my other clients. I was planning to set up a new 2.4Ghz wireless client interface on one of my OpenWrt-based APs, specify the SSID/WPA key of the deauther, and route the IP address of it to my home network ones. Or just set a static interface IP for the OpenWrt device within the same range (192.168.4.100/24).

_By the way, is the dhcp lease given out by the deauther is a /24 range? Or /32? Or... the question here is, multiple devices could connect to it using the default IP setup?_

As I have seen in the OpenWrt logs (and on it's web inteface) that it is trying to connect, but it disconnects endlessly with the below error messages:

_(Reason: 3=DEAUTH_LEAVING)_

```
Tue Jul 28 12:45:26 2020 kern.info kernel: [675929.679218] ap1-2g-pwn: authenticate with 00:0f:b4:d1:61:c6
Tue Jul 28 12:45:26 2020 kern.info kernel: [675929.699019] ap1-2g-pwn: send auth to 00:0f:b4:d1:61:c6 (try 1/3)
Tue Jul 28 12:45:26 2020 kern.info kernel: [675929.708006] ap1-2g-pwn: authenticated
Tue Jul 28 12:45:26 2020 kern.info kernel: [675929.713942] ath9k 18100000.wmac ap1-2g-pwn: disabling HT/VHT/HE as WMM/QoS is not supported by the AP
Tue Jul 28 12:45:26 2020 kern.info kernel: [675929.726079] ap1-2g-pwn: associate with 00:0f:b4:d1:61:c6 (try 1/3)
Tue Jul 28 12:45:26 2020 kern.info kernel: [675929.735037] ap1-2g-pwn: RX AssocResp from 00:0f:b4:d1:61:c6 (capab=0x11 status=0 aid=1)
Tue Jul 28 12:45:26 2020 kern.info kernel: [675929.743690] ap1-2g-pwn: associated
Tue Jul 28 12:45:26 2020 daemon.notice netifd: Network device 'ap1-2g-pwn' link is up
Tue Jul 28 12:45:26 2020 daemon.notice netifd: Interface 'pwn' has link connectivity
Tue Jul 28 12:45:36 2020 kern.info kernel: [675939.753327] ap1-2g-pwn: deauthenticating from 00:0f:b4:d1:61:c6 by local choice (Reason: 3=DEAUTH_LEAVING)
Tue Jul 28 12:45:36 2020 daemon.notice netifd: Network device 'ap1-2g-pwn' link is down
Tue Jul 28 12:45:36 2020 daemon.notice netifd: Interface 'pwn' has link connectivity loss
Tue Jul 28 12:50:37 2020 kern.info kernel: [676240.631015] ap1-2g-pwn: authenticate with 00:0f:b4:d1:61:c6
Tue Jul 28 12:50:37 2020 kern.info kernel: [676240.650456] ap1-2g-pwn: send auth to 00:0f:b4:d1:61:c6 (try 1/3)
Tue Jul 28 12:50:37 2020 kern.info kernel: [676240.658605] ap1-2g-pwn: authenticated
Tue Jul 28 12:50:37 2020 kern.info kernel: [676240.664725] ath9k 18100000.wmac ap1-2g-pwn: disabling HT/VHT/HE as WMM/QoS is not supported by the AP
Tue Jul 28 12:50:37 2020 kern.info kernel: [676240.678185] ap1-2g-pwn: associate with 00:0f:b4:d1:61:c6 (try 1/3)
Tue Jul 28 12:50:37 2020 kern.info kernel: [676240.687069] ap1-2g-pwn: RX AssocResp from 00:0f:b4:d1:61:c6 (capab=0x11 status=0 aid=1)
Tue Jul 28 12:50:37 2020 kern.info kernel: [676240.696859] ap1-2g-pwn: associated
Tue Jul 28 12:50:37 2020 daemon.notice netifd: Network device 'ap1-2g-pwn' link is up
Tue Jul 28 12:50:37 2020 daemon.notice netifd: Interface 'pwn' has link connectivity
Tue Jul 28 12:50:47 2020 kern.info kernel: [676250.710108] ap1-2g-pwn: deauthenticating from 00:0f:b4:d1:61:c6 by local choice (Reason: 3=DEAUTH_LEAVING)
Tue Jul 28 12:50:47 2020 daemon.notice netifd: Network device 'ap1-2g-pwn' link is down
Tue Jul 28 12:50:47 2020 daemon.notice netifd: Interface 'pwn' has link connectivity loss
```

**To Reproduce**
Steps to reproduce the behavior:

1. OpenWRT - > go to wireless settings, Scan for new networks under your 2.4Ghz's radio settings
2. Connect with your deauther's SSID, specifiy the password, create a new network interface, and leave DHCP protocol or add a static ip for the same subnet as the deauther (192.168.4.x)
3. Save settings, check in OpenWrt if the interface is connected.
4. See error in openwrt syslog

**Expected behavior**

OpenWrt should connect to the deauther and get an IP or establish connection with deauther as any other wireless client, eg. Windows PC

**Environment (please complete the following information):**
 - OS: OpenWrt SNAPSHOT
 - Browser: Firefox
 - Version2.50 RC
 - Hardware: new NoceMcu v3, aka. chinese clone from Aliexpress

**Attempts**
Tried deauther v2.10 and 2.50 RC

**Additional context**
Deauther is working as expected by directly connecting to it with another wireless client, eg. Windows 10 PC and USB Wi-Fi adapter.

eisengrau, 2020-07-29:
Ok, it seems like the issue was, that I have the deauther's SSID to hidden, and, seemingly OpenWrt didn't like that, even if I added the correct ESSID and BSSID of the deauther.

On a further note, is there a possibility to change the mode on how the deauther connects to a client? Eg. it is possible to specify in the config that the deauther should connect to an AP as a client, instead the other way around?

spacehuhn, 2020-07-29:
> On a further note, is there a possibility to change the mode on how the deauther connects to a client? Eg. it is possible to specify in the config that the deauther should connect to an AP as a client, instead the other way around?

Currently not, but something I will have a look at for version 3.0 🙂

closed 2020-11-03 by stale[bot]