SpacehuhnTech/esp8266_deauther #1285: Can't connect to pwned (ssid)

rexsllemel, 2020-07-28:
> Have you searched for existing (open and closed) issues describing the same problem?
Yes, and they have different issue than mine I guess? they solved their issue but can't fix mine the way they fix their issues.

**Describe the error**
A clear and concise description of what the problem is.

 I downloaded/cloned the deauther, extract it and on the folder esp8266_deauther-2  I navigate to esp8266_deauther and use the esp8266_deauther.ino (arduino). I compiled/check it first and there seems to be no error ( I already installed the prerequisites on the board manager). The flashing is successful. I can see the ssid (pwned) but did not connect to it yet, I restart the NodeMCU after a while and then trying to connect with the password of deauther but it says "incorrect password" so  I decided to reflash again and again.

What do you think causes it?

Maybe because this is a cheap NodeMCU V3? a cloned one.

Error/Compile/Output Log

This is what showing on serial monitor (NOT HELPFUL ENOUGH only showing a square box looping together with the flashing blue light 3 times)

⸮
ȌΌ⸮⸮⸮⸮!⸮J⸮⸮JH⸮@⸮ǝ⸮⸮⸮⸮
(sample only because I can't paste more of them)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Phone settings
2. Click on set to statip ip (192.168.4.1)
3. Connect to it
4. See error

**Expected behavior**

Connect to pwned(ssid) normally/successfully and use the web interface.

**Screenshots**

[image]

**Environment (please complete the following information):**
 - OS: Windows 7x64
 - Browser: Safari
 - Version: 2? (deauther version 2)
 - Hardware: esp8266 NodeMCU v3

**Attempts**
What have you already tried and didn't work out.

Changing the default password. (not working)
Changing the default ssid from "pwned" to "Amazing". (not working)
Removing the password to make it open. (not working)

**Additional context**
Provide as much information as possible, better too much than too little!

I removed the password and connect my iphone 5s and it say's "unable to connect" so I decided to input static ip address 192.168.4.25 255.255.255.0 192.168.4.1 but still unable to connect so I restarted the nodemcu, turn off the wifi and turn it on again and now it connect but it disconnect automatically for  only 1 second.

So I am guessing here that the nodemcu keep restarting every 2 seconds? some internal error? as you can see the screenshot it loops like that and the blue light flashing 3 times again and again. They way the blue light flashes, the log on serial monitor filled with text (square box) 3 TIMES.

 I hope anyone could help me.

If you don't use this template, your issue might be closed and tagged invalid!
I do because this is rare issue. Thank you!

rexsllemel, 2020-07-28:
Update:

On nodeMCU firmware programmer works fine.

a5tabout, 2020-08-26:
@rexsllemel You have to change the baudrate (speed at which data is sent) of the Serial communication,
In the screenshot you have given, at the bottom-right of the window you can see **9600 baud**
change it to **115200 baud** and you should have readable output

rexsllemel, 2020-09-03:
> @rexsllemel You have to change the baudrate (speed at which data is sent) of the Serial communication,
> In the screenshot you have given, at the bottom-right of the window you can see **9600 baud**
> change it to **115200 baud** and you should have readable output

thanks! now I understand some things

Driv28, 2020-09-17:
Do you have a solution? Same problem here

a5tabout, 2020-09-24:
@Driv28 Try commenting one of the save methods at the end of the .ino file and reupload
I remember one of them was causing m,e this same error but I cannot remember which one it was.

sedaf5, 2020-09-26:
I also have this problems @Driv28 so can @spacehuhn  can  you fix this ?

GrandJohnny, 2020-10-12:
> **Describe the error**
> A clear and concise description of what the problem is.
>
> I downloaded/cloned the deauther, extract it and on the folder esp8266_deauther-2 I navigate to esp8266_deauther and [...]

i have the same problem with the same serial output shown in that screenshot
is there any fix?

mnemocron, 2020-11-14:
Same issue here on nodemcu 0.9
Arduino 1.8.12

```
Mounting SPIFFS...OK
Loading settings...OK
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: true
STARTED! \o/
2.2.0
# scan -t 5s
Stopped scan
Scan results saved in /scan.json
Removed all APs
Cleared station list
Starting scan for access points (Wi-Fi networks)...

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4026fb37 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffcd0 end: 3fffffc0 offset: 01a0
3ffffe70:  3ffe8898 4026fb37 3fffff70 40212770  
3ffffe80:  2e322e00 006e0030 80000000 402255a7  
3ffffe90:  3fff11e1 3ffeec5c 3fffff70 40212a08  
3ffffea0:  3fff11e1 3ffeec5c 3ffeed10 40212e59  
3ffffeb0:  00000202 00000001 0000003c 00000000  
3ffffec0:  00000258 00000119 00000001 00000001  
3ffffed0:  381d0001 00f7b0f9 484e800e 00c80004  
3ffffee0:  77700003 0064656e 00000000 00000000  
3ffffef0:  00000000 00000000 00000000 00000000  
3fffff00:  64000000 74756165 00726568 00000000  
3fffff10:  00000000 00000000 00000000 00000000  
3fffff20:  00000000 00000000 00000000 00000000  
3fffff30:  00000000 00000000 00000000 00000000  
3fffff40:  00000000 04a8c000 00010101 01006e65  
3fffff50:  00000101 00000000 00000258 1db36daa  
3fffff60:  29609106 67f6300f 18b2956e 6a1994a6  
3fffff70:  3fff3324 0014025f 00fef01c 3ffee740  
3fffff80:  000001e9 3ffee73c 3ffeec5c 4021a644  
3fffff90:  feefeffe feefeffe feefeffe 3ffef05c  
3fffffa0:  3fffdad0 00000000 3ffef01c 402267c8  
3fffffb0:  feefeffe feefeffe 3ffe8540 4010119d  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
```

As mentioned by @a5tabout 
commenting the lines 

```c
        //names.save(false);
        //ssids.save(false);
        //settings.save(false);
```

worked

closed 2021-02-19 by stale[bot]