SpacehuhnTech/esp8266_deauther #1288: Adding captive portal for phishing

joldboy69, 2020-08-04:
**Do similar feature requests issues already exist (open and closed)?**
No

**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I would like a feature where the attacker has the ability to create an evil twin (like in the beacon spammer feature) and be able to have a victim connect to the evil twin network and be prompted for a password to put in (which is sent to the attacker).

**Describe alternatives you've considered**
There are other github repos that can create a captive portal but cannot also deauth another network at the same time. 

**Additional context**
No

Bommlon, 2020-08-06:
This should already be possible using Deauther v3 and Huhnitor.
I'm not sure about the password tho.

vuongfcc, 2020-11-03:
This is what you want https://www.hackster.io/234v244/esp8266-wifiphisher-4ed3d3

joldboy69, 2020-11-18:
Okay everybody. I actually forgot to post this but about 2 weeks ago, I was able to combine this project with another esp8266 project dedicated to phishing. (https://github.com/125K/ESP8266_WiFi_Captive_Portal). When I can, I will edit this comment to give out the source code + flashable binary.

cagdasb, 2020-11-19:
> 
> 
> Okay everybody. I actually forgot to post this but about 2 weeks ago, I was able to combine this project with another esp8266 project dedicated to phishing. (https://github.com/125K/ESP8266_WiFi_Captive_Portal). When I can, I will edit this comment to give out the source code + flashable binary.

Man you are awesome. I'm dying to find something like that. 
So what it does is simply use this to deauth "https://github.com/SpacehuhnTech/esp8266_deauther"
Use this to open captive portal "https://github.com/125K/ESP8266_WiFi_Captive_Portal" right?

That would be awesome if it start blinking and doesn't stop until we stop deauth. And wifi-captive-portal project doesn't save passwords. I would be glad if you can find a way to do that.

I'm looking for your reply. Thank you

joldboy69, 2020-11-20:
Thanks! Unfortunately, i accidentally deleted the modified code so i had to rebuild it :(

Even then, im not sure if i put everything back together as i remember. But anyways, if you want the source code, here ya go:

https://filebin.net/bml1us2qkk213dcl

(keep in mind that you may have to modify the code yourself in case you're using a different board or something. Im actually using the dstike deauther watch v2.)

cagdasb, 2020-11-20:
> Thanks! Unfortunately, i accidentally deleted the modified code so i had to rebuild it :(
>
> Even then, im not sure if i put everything back together as i remember. But anyways, if you want the source [...]

I fucking love you man :D 
Thansk for the effort but I have no idea about coding.. I'm using NODEMCU board and I don't know how to compile it either.
Could you spear a little time for it? That's really important for me.

joldboy69, 2020-11-22:
Either you will need to change the board you are using the arduino ide or i made a mistake in the code. If you're getting an error, what does it say?

cagdasb, 2020-11-22:
Man I have no idea about how does "the arduino ide" works or about coding. Could you please uplaod it as bin file?

joldboy69, 2020-11-25:
Sorry for the late response. Fortunately, im finally on break from school so i can get back to it :D

joldboy69, 2020-11-25:
Okay. This should do the job. I havent tested it but you should be able to access the phishing panel with "192.168.4.1" and the attack panel with "172.2.1.1" in your url bar. https://filebin.net/ci0uhxhta8gcmxjz

tqcab, 2020-11-25:
> Cảm ơn! Thật không may, tôi đã vô tình xóa mã đã sửa đổi nên tôi phải xây dựng lại nó :(
> 
> Ngay cả khi đó, tôi không chắc liệu mình có đặt mọi thứ lại với nhau như tôi nhớ hay không. Nhưng dù sao đi nữa, nếu bạn muốn mã nguồn, đây là:
> 
> https://filebin.net/bml1us2qkk213dcl
> 
> (Hãy nhớ rằng bạn có thể phải tự sửa đổi mã trong trường hợp bạn đang sử dụng bảng khác hoặc thứ gì đó. Tôi thực sự đang sử dụng đồng hồ dstike deauther v2.)

I got this message while compiling it using arduinno

```
Arduino: 1.8.13 (Windows 7), Board: "NodeMCU, 4MB (FS:1MB OTA:~1019KB), Only Sketch, 115200"

c:/users/vuonghan/appdata/local/arduino15/packages/deauther/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: sketch\Attack.cpp.o:(.text._ZN6AttackC2Ev+0x10): undefined reference to `currentTime'

c:/users/vuonghan/appdata/local/arduino15/packages/deauther/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: sketch\CLI.cpp.o:(.text._ZN3CLI10runCommandE6String+0x2f0): undefined reference to `autosaveTime'

collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "SSD1306Wire.h"

 Used: C:\Users\VuongHan\Documents\Arduino\libraries\SSD1306

 Not used: C:\Users\VuongHan\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.7.2\libraries\esp8266-oled-ssd1306

exit status 1

Error compiling for board NodeMCU.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

joldboy69, 2020-11-25:
Yea. I had to fix that. Have you tried flashing the compiled binary? Here is the fixed code btw https://filebin.net/vl4l9p3q7q6dm1bp

tqcab, 2020-11-25:
> Yea. I had to fix that. Have you tried flashing the compiled binary? Here is the fixed code btw https://filebin.net/vl4l9p3q7q6dm1bp

thanks, i have tried the binary code, and cannot access 172.2.1.1

joldboy69, 2020-11-25:
Oh that sucks. Can you access the phishing portal? I originally made this for the deauth watch. Ill keep working at it though.

cagdasb, 2020-11-25:
> 
> 
> Okay. This should do the job. I havent tested it but you should be able to access the phishing panel with "192.168.4.1" and the attack panel with "172.2.1.1" in your url bar. https://filebin.net/ci0uhxhta8gcmxjz

Hello again @joldboy69 
I just Flashed it to Nodemcu ESP8266 board.
I can access to phishing portal 192.168.4.1 but I can't access to deauth portal 172.2.1.1
I can try flashing it again if you can modify the bin file. Thanks again.

tqcab, 2020-11-25:
> Oh that sucks. Can you access the phishing portal (192.168.4.1)? I originally made this for the deauth watch. Ill keep working at it though. Im gonna take a guess and say that the board cant run two servers at once.

i tried with d1 mini, nodemcu, esp07,
Try binary code, try arduino. but neither of them can connect to 172.2.2.1 or 172.2.1.1
I can only access
192.168.4.1

joldboy69, 2020-11-25:
As i said, i dont think the arduino module can run two servers at once.

AnonDXT-XcX31, 2020-11-27:
hey guys 1st you both are amazing and your thinkings are next level !!!!!!!
but my question is if we use two different esp8266 one for deauth and one for captive portal ?? is int that fine like what we do in a kali linux use two wifi cards to attack one for capture and one for deauth !!!!

cagdasb, 2020-11-27:
> 
> 
> hey guys 1st you both are amazing and your thinkings are next level !!!!!!!
> but my question is if we use two different esp8266 one for deauth and one for captive portal ?? is int that fine like what we do in a kali linux use two wifi cards to attack one for capture and one for deauth !!!!

Hey, I do the samething on kali too. Yes it's fine you can do that.

joldboy69, 2021-01-17:
Wow. Been a while. So it appears that the newest version of the project (version 3) WILL have a rogue ap to connect to. Hasn't been setup for the deauther watch yet, so it might be a few weeks or months until the next big thing

ngocthuong234, 2021-08-17:
With only one esp8266, deauth + wifi password + Facebook, Google, Twitter,..
https://youtu.be/oKsRb4KXXk0
https://youtu.be/Nt-0aTfPECQ

Crossik9, 2021-11-16:
> 
> 
> Wow. Been a while. So it appears that the newest version of the project (version 3) WILL have a rogue ap to connect to. Hasn't been setup for the deauther watch yet, so it might be a few weeks or months until the next big thing

Hi m8, I would love to see your work, however, links are dead. Are you still have source code around?