SpacehuhnTech/esp8266_deauther #1290: adding Evil twin

adrianlol2017, 2020-08-17:
hi, is it possible that we can have an evil twin attack function because I think it is possible but I am not sure but if possible you can add the function?

syssytheboss, 2020-08-17:
Yeah, this would be a good thing to have.It may require 2 devices: One will keep deauthing while the other try to capture the handshake and start an AP.A communication is optional but great to have between the two devices(maybe BLT or other RF systems) in order to stop the deauth process, but you can always switch the deauth device off after you got the pass.

ngocthuong234, 2021-08-17:
With only one esp8266, deauth + wifi password + Facebook, Google, Twitter,..
https://youtu.be/oKsRb4KXXk0
https://youtu.be/Nt-0aTfPECQ

anonymousseafarer, 2021-08-17:
> With only one esp8266, deauth + wifi password + Facebook, Google, Twitter,..
> https://youtu.be/oKsRb4KXXk0
> https://youtu.be/Nt-0aTfPECQ

> With only one esp8266, deauth + wifi password + Facebook, Google, Twitter,..
> https://youtu.be/oKsRb4KXXk0
> https://youtu.be/Nt-0aTfPECQ

But your firmware is not english.. can you share the bin file? Or just want money for it. Please share it for free.. thank you.