SpacehuhnTech/esp8266_deauther #1299: Deauther can't Attack Laptop near to the Wi-Fi

yamyamii, 2020-09-06:
Hey! , I have already setup my device with everything completed. Yesterday i have tried the device in my friends cafe , he plays music it is near the router with no ethernet cable plugged on the laptop. When i start deauther Attack it makes devices on cafe disconnect from the wifi , but it somehow doesnt effect the laptop that is near the wifi. It is still connected. I have placed deauther device near the laptop but still it didn't effect it. I have tried Beacon (clone) , It works but the device cant make laptop disconnect. Is it because the laptop being closer to router or am i have to setup my deauther again?

yamyamii, 2020-09-06:
btw the wifi i am attacking is 2.4 ghz

Bommlon, 2020-09-12:
For legal reasons, please make sure you have his permission to use your device on his network.
The distance to the router does not affect if you are able to deauth a device as long as you are connected. Maybe there are other aspects you didn't take into account (Laptop is on another network or not connected at all).

closed 2020-12-18 by stale[bot]