SpacehuhnTech/esp8266_deauther #1301: Deauthuer 2.5 Users not showing

PumpP, 2020-09-09:
Hello all, I'm new to this and I have tested my first projet which is the deauther version 2.5, all seems to work okay but I cannot see users connected to my network, I have seen a screenshot of this same program that had a user tab at the top next to scan.
Have I missed something?

AleksZavg, 2020-09-10:
Try installing this version v2. 1. 0
https://github.com/SpacehuhnTech/esp8266_deauther/releases/tag/v2.1.0

PumpP, 2020-09-10:
Many thanks.

Kind regards.

Pete.

The information contained in this e-mail is confidential and may be legally privileged, except where expressly indicated, it may not necessarily represent my views.
This e-mail is intended solely for the addressee(s) and I ask that any unauthorised recipient advise the sender immediately. If you are not the intended recipient any disclosure, copying, distribution or any other action taken in reliance on it is prohibited and may be unlawful.

vovanhau, 2020-09-12:
> luôn sẵn lòng giúp đỡ

Is there a way to attack a device on boot by mac address without selecting it from a scan?

AleksZavg, 2020-09-12:
After the first scan, you can save your device or wi-fi network to esp8266, and after rebooting, you can immediately select the device you saved! You can also change the AutoSave time in the settings:)

vovanhau, 2020-09-12:
https://www.youtube.com/watch?v=HWZrBoIKacw
Is it possible to do so on 8266 mcu modems or not

AleksZavg, 2020-09-12:
theoretically esp8266 can attack everyone at once, but the current firmware does not have such a function.  As for a MAC attack without an initial scan, I'm not sure if this can be done.  If you need to disable everyone at once, that is, mdk4 on github;)

Letschat44, 2020-09-12:
Stop email me

Bommlon, 2020-09-14:
@Letschat44 just unsubscribe

Letschat44, 2020-09-14:
How do I unsubscribe so I don't get 400 emails ￼at ?

closed 2020-12-20 by stale[bot]