SpacehuhnTech/esp8266_deauther #1302: No such File or Directory for <LittleFS.h>

Damocles6691, 2020-09-10:
This is the first time i'm using Arduino and I keep running into this issue, and I don't know how to fix it.

Arduino: 1.8.13 (Mac OS X), Board: "Generic ESP8266 Module, Serial, 80 MHz, 80MHz, DIO, 115200, 512K (64K SPIFFS), ck"

In file included from /Users/(name)/Desktop/esp8266_deauther-2.5.0-rc2/esp8266_deauther/esp8266_deauther.ino:27:0:
functions.h:5:22: fatal error: LittleFS.h: No such file or directory
 #include <LittleFS.h>
                      ^
compilation terminated.
exit status 1
LittleFS.h: No such file or directory
Index error: could not find referenced tool name=mkspiffs version=0.1.2 packager=esp8266
Index error: could not find referenced tool name=esptool version=0.4.9 packager=esp8266
Index error: could not find referenced tool name=xtensa-lx106-elf-gcc version=1.20.0-26-gb404fb9-2 packager=esp8266

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

i'm running this on a MacOS Catalina on 10.15.3, It's i've gotten all of my code off of the Chrome browser, and i've already had to get a CP210x driver. I've followed all the steps from the tutorial, i'm confident it's going to the right port and into the right Board. My guess was I don't have the "LittleFS.h" file in the library; I looked and it wasn't there, so i'm lost. I don't know if this is enough information to go off of, but this is the best I can think of to help me with troubleshooting.

Maschine2501, 2020-09-15:
Same Issue here.
Generic ESP8266 / Bordlibrary 2.0.0

spacehuhn, 2020-09-16:
We have a new Arduino board manager URL
`https://raw.githubusercontent.com/SpacehuhnTech/arduino/main/package_spacehuhn_index.json`.
Try uninstalling the current deauther boards and installing the latest one, as described [here](https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide)

Damocles6691, 2020-09-17:
Now I keep getting stuck with this problem when verifying

fork/exec /Users/(Name)/Library/Arduino15/packages/deauther/tools/python3/3.7.2-post1/python3: no such file or directory
Error compiling for board Generic ESP8266.

closed 2020-12-24 by stale[bot]