SpacehuhnTech/esp8266_deauther #1304: Heltec esp8266

GrandJohnny, 2020-09-12:
> Have you searched for existing (open and closed) issues describing the same problem?

yeah but found nothing...

**Describe the error**
i got a wifi kit 8 (heltec) which is a esp8266 board with an 0.91 oled on it. i could get it to work ( with display ) by editing things that i dont remember now because i lost my laptop 3 weeks ago and all i remember is that i edited A_config.h and also added some of heltec original libraries for this board ... so anyways i tried many other oled libraries on this but im getting a black screen with nothing ( but oled works properly with other scripts )
and one thing i know is that it is required to set the OLED RST like this: (which i found it in heltec provided libraries )
 SSD1306Wire(0x3c, SDA, SCL, OLED_RST, GEOMETRY_128_32);

but i cant remember how i used this :(

 - Hardware : heltec esp8266 oled 0.91 ( wifi kit 8 )

GrandJohnny, 2020-09-12:
the problem seems to be the new updates i mean i can see under ....\Arduino15\packages\deauther\hardware\esp8266 that there are too versions which when i updated my arduino ide its recently added and i need to know how can i set the old libraries as default?

Bommlon, 2020-09-14:
you should try it out

GrandJohnny, 2020-09-14:
so i fixed it and im going to upload those changes for anyone who is wondering and needs it on a 0.91 OLED

GrandJohnny, 2020-09-14:
so the working and adjusted code is here :
https://github.com/GrandJohnny/Heltec-deauther
hope it help someone

_Happy Deauthing :)_

closed 2020-09-14 by GrandJohnny