SpacehuhnTech/esp8266_deauther #1305: Would it be possible to share source code for esp8266_deauther_2.5.0-rc2_DSTIKE_DEAUTHER_WRISTBAND.bin

Thman558, 2020-09-14:
Hi! I was just wondering if @spacehuhn would let me download the source code for the rc2 DSTIKE_DEAUTHER wristband it comes in a .bin file and I would like a .ino so I can see it in Arduino ide? I tried using the source code at the bottom but when I flashed it my led nor screen would turn on. Any help Thanks!

AleksZavg, 2020-09-14:
At the bottom of the repository https://github.com/SpacehuhnTech/esp8266_deauther there is an MIT License. It says what can be done with its repository in terms of copyright. I recommend reading it anyway

Thman558, 2020-09-14:
Wasn't exactly what I was looking for I just wanted the rc2 DSTIKE_DEAUTHER WRISTBAND.bin in .ino format Still interesting though!

satan666000, 2020-09-23:
En Arduino igual menda error al flashear y no enciende por eso te recomiendo que uses tu teléfono android y así es más facil

satan666000, 2020-09-23:
In Arduino the same error when flashing and it does not turn on, so I recommend that you use your android phone and that is easier

closed 2020-12-29 by stale[bot]