SpacehuhnTech/esp8266_deauther #1307: Failed to connect to esp8266 board with arudino IDE

raccoondude, 2020-09-16:
I attempted to connect my board with the arudino IDE, apon flashing I got this on the terminal

```
esptool.py v2.8
Serial port /dev/ttyS0
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "/root/.arduino15/packages/deauther/hardware/esp8266/2.7.2/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "/root/.arduino15/packages/deauther/hardware/esp8266/2.7.2/tools/esptool/esptool.py", line 2890, in main
    esp.connect(args.before)
  File "/root/.arduino15/packages/deauther/hardware/esp8266/2.7.2/tools/esptool/esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
```

Any clue on how to fix this? Im not sure how im supposed to connect it!

AleksZavg, 2020-09-17:
I had the same problem, you need to hold down the FLASH button on esp8266 attach the Board to the Arduino and start compiling and downloading, and as soon as the firmware percentages go in the Arduino download window, you can release the FLASH button, but you can't release the button until the firmware has

satan666000, 2020-09-23:
if what you are looking for is to flash the .good with an android phone it is very easy

raccoondude, 2020-10-01:
I have no flash button, I don't see it atleast

AleksZavg, 2020-10-01:
If you don't have this button, I don't know what else can help you! Try to find out which output this button was connected to and make your own (short-circuit to ground),
[image]
[image]

closed 2021-01-06 by stale[bot]