SpacehuhnTech/esp8266_deauther #1308: How Can I jamming all wifi automatic ?

muratoznn, 2020-09-17:
I don't want wifi to be used in the house I'm in. I cannot keep up with this. How can I make the nodemcu development board automatically find and close all wifi connections? This means the following. I will not deal with it, it will take care of it. Does this device have such a setting? Can Nodemcu close all wifi connections automatically ?

muratoznn, 2020-09-17:
I looked but I cant find please answer my question ?

AleksZavg, 2020-09-17:
If I understand correctly, you need esp8266 to attack everything and everyone. ESP8266 can't do that! But there is a program mdk4 here it can, but it needs a special wifi adapter and driver on kali linux!

muratoznn, 2020-09-17:
I want to understand that? If I save some wifi names. Then when the nodemcu runs the development board, can it cut only those wifi names automatically? I mean this. When I save some wifi names on the device and then run it, can only that wifies stop communication? So can I use this device for certain wifies automatically?  Or will the device stop when the wifi interface connection is lost?

muratoznn, 2020-09-17:
I want to save some wifi names and I want to jamming only these wifi names automatically ? Can be do this or will the device stop when the wifi interface 192.168.0.4 connection is lost ?

AleksZavg, 2020-09-18:
1) you can run a wifi scan and save the name and MAC of the wifi access point or client to the non-volatile memory of the esp8266. This data will be stored there until you delete it! And the next time you launch it, you can immediately select the access point you saved and immediately launch an attack without scanning. And if such a point is not nearby, then esp8266 will still attack it regardless of whether the network is on the air or not.
2) in the esp8266 settings, there is a time setting after which the attack will stop, if you want the attack to not stop, then change this parameter to 0 (it will say)

muratoznn, 2020-09-18:
OKay that means this. I can manually activate this device with the help of a battery, then set it to work continuously and disconnect the web interface? Did I get right ?

muratoznn, 2020-09-18:
I will activate this device near my girlfriend's house with the help of the battery and then run away. I need help for this. So I'm going to do a small James Bond operation. After I activate the device, I have to get away from it. It is enough that it does not close after I leave. I guess it can do so much

Bommlon, 2020-09-18:
I'd like to inform you that depending on the laws in your country this may be illegal. Besides that, Spacehuhn Technologies does not endorse using our software against others.

muratoznn, 2020-09-18:
I know it . This is illegal. This crime has a fine. And I'm willing to pay this money to this person

AleksZavg, 2020-09-19:
If you are going to use a network cloning attack and manage esp8266 via the web, I recommend that you do not use these settings :
1) Beaconchannel
2) Beaconinterval
Because after launching the attack, you will lose your wifi connection with esp8266, and you will no longer be able to control it

spacehuhn, 2020-09-19:
> I know it . This is illegal. This crime has a fine. And I'm willing to pay this money to this person

No this is not like a parking ticket. You can harm highly important infrastructure unintentionally, because you have no control over the targeted networks anymore. So besides all the technical problems to achieve such a functionality, it's not ethical and not what this community or this project stands for!
You are free to do whatever you want **on your own** and bare the responsibility for that. But I do not want my work or its community associated with your potential criminal activity.

closed 2020-09-19 by spacehuhn