SpacehuhnTech/esp8266_deauther #1313: Deauther esp32

Edo-Bat, 2020-09-24:
Hi, can i use the code for esp8266 deauther also on esp32?

AleksZavg, 2020-09-28:
The ESP32 is not supported this firmware!
[image]

Edo-Bat, 2020-09-29:
Ok : (  How could I use the same program on the 32?

AleksZavg, 2020-09-29:
I don't know, but I think you need to rewrite the code to esp32. You may need to use libraries for esp32 in your code.

closed 2021-01-05 by stale[bot]

spacehuhn, 2021-12-12:
Hi! Since there were so many issues on this repo about supporting the ESP32, I published a blog post to bring some clarity to this topic: https://blog.spacehuhn.com/esp32-deauther/ 
TL;DR: It's made for the ESP8266, and I have no plans to port this application to the ESP32. But it's open-source, so you're free to do it yourself if you want.