SpacehuhnTech/esp8266_deauther #1314: Next Page for extra options in Display

GrandJohnny, 2020-09-24:
hello there
how can we define if there is more menu nodes so it goes to next page?
for example if i add a new option on the menu so they become 6 options but only 5 can be displayed (while my oled is big enough and configured if there would be more content ) ?

i can see there are somethings about this in void DisplayUI::drawMenu()
but cant figure which part is designed for this ( going to next page if there are more options than 5 )

int row = (currentMenu->selected / 4) * 5; // /5

i edited the above value to 4 to check if the clock would go to next page and it kinda works but the problem is that when it goes to next page its blank and i cant see "CLOCK" option

and the real situation is that im configuring this on a 0.91 OLED device and the last unsolved thing about is that i need to force it go to the next page after 4 options but i dont know where this is defined in your code

AleksZavg, 2020-09-28:
Hi! As far as I understand you don't know how the display will behave if there are a lot of items. Since I don't know the Arduino programming language, but I think this will help you https://www.youtube.com/watch?v=N0s2Qmrsu4A .
This video is in Russian, but I think it will still help you!

GrandJohnny, 2020-09-28:
> 
> 
> Hi! As far as I understand you don't know how the display will behave if there are a lot of items. Since I don't know the Arduino programming language, but I think this will help you https://www.youtube.com/watch?v=N0s2Qmrsu4A .
> This video is in Russian, but I think it will still help you!

Thanks that video really helped me out to figure whats really going on an oled

i also figured out how we can define how many options we want to be displayed by adjusting the following part in DisplayUI.cpp

`void DisplayUI::drawMenu() {
    String tmp;
    int    tmpLen;
    int    row = (currentMenu->selected / 4) * 4; // /5 * 5

    // correct selected if it's off
    if (currentMenu->selected < 0) currentMenu->selected = 0;
    else if (currentMenu->selected >= currentMenu->list->size()) currentMenu->selected = currentMenu->list->size() - 1;

    // draw menu entries
    for (int i = row; i < currentMenu->list->size() && i < row + 4; i++) { // + 5 
        tmp    = currentMenu->list->get(i).getStr();
        tmpLen = tmp.length();

        // horizontal scrolling
        if ((currentMenu->selected == i) && (tmpLen >= maxLen)) {
            tmp = tmp + tmp;
            tmp = tmp.substring(scrollCounter, scrollCounter + maxLen - 1);
            
            if ((scrollCounter > 0 && scrollTime < currentTime - scrollSpeed) || (scrollCounter == 0 && scrollTime < currentTime - scrollSpeed * 4)){
              scrollTime = currentTime;
              scrollCounter++;
            }
            
            if (scrollCounter > tmpLen) scrollCounter = 0;
        }

        tmp = (currentMenu->selected == i ? CURSOR : SPACE) + tmp;
        drawString(0, (i - row) * 8, tmp); //
    }
}`

- change any "4" you see in above code which in above adjusted code the menu contains only 4 options and will go to the next page if you try to select 5th option and etc ...

btw i needed this for my 0.91 oled heltec device which all adjusted code is available code >[here](https://github.com/GrandJohnny/Heltec-deauther)<

closed 2020-09-28 by GrandJohnny