SpacehuhnTech/esp8266_deauther #1319: Can this be used to help me pentest WiFi passwords?

Thman558, 2020-10-01:
Hello, I was wondering if this could be used with my computer while it is packet capturing and I deauth someone with it will my computer catch the 2 way handshake? Thanks

AleksZavg, 2020-10-01:
this device sends deauthentication packets that disconnect a particular client from the network! And you can't know exactly how many packets she sent. Using esp8266 to capture a handshake is inconvenient!
It will be easier to use special programs for this purpose (wifite, airmon-ng, mdk4)

Thman558, 2020-10-02:
I know I normally use aircraft-ng just thought this would be a cool tool to help me

AleksZavg, 2020-10-02:
esp8266 perfectly replaces mdk4! You can use it for these purposes!
But my opinion is that it is better to use special programs, because if you send too many deauntification packets, the device will stop trying to connect to the network again! Thinking that it is unavailable!

closed 2021-01-07 by stale[bot]