SpacehuhnTech/esp8266_deauther #1321: How to add more than 60 networks to the beacon?

adiamaq, 2020-10-03:
60 seems to be the limit of fake flood ssids, no matter what (list goes from 0 to 59). I tried editing a few values in for loops that went to 60 but that is only for displaying. I also tried opening the files, the function that adds the ssid's use getFile(run?cmd "command") but I still can't find where the hardcoded limit is. Cheers :)

closed 2021-01-08 by stale[bot]

spacehuhn, 2021-01-11:
There has to be a limit to make sure there's enough RAM left for the ESP8266 to function properly. But I made a spereate project for this a while ago without that limitation: https://github.com/spacehuhn/esp8266_beaconSpam/

adiamaq, 2021-01-11:
Awesome, thank you so much, for replying and for your work! Will use it very soon. Cheers!