SpacehuhnTech/esp8266_deauther #1325: What Version of ArduinoJSON to install

Thman558, 2020-10-08:
Hello, I am trying to compile in and I keep getting this error `In file included from /Users/rileynichols/Library/Arduino15/packages/deauther/hardware/esp8266/2.7.2/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6:0,
                 from /Users/rileynichols/Library/Arduino15/packages/deauther/hardware/esp8266/2.7.2/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,
                 from /Users/rileynichols/Library/Arduino15/packages/deauther/hardware/esp8266/2.7.2/cores/esp8266/Arduino.h:32,
                 from sketch/esp8266_deauther.ino.cpp:1:
/Users/rileynichols/Library/Arduino15/packages/deauther/hardware/esp8266/2.7.2/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:25:130: error: redefinition of 'const char S_JSON_DISPLAYINTERFACE []'
   #define PROGMEM __attribute__((section( "\".irom.text." __FILE__ "." __STRINGIZE(__LINE__) "."  __STRINGIZE(__COUNTER__) "\"")))
                                                                                                                                  ^
sketch/settings.h:81:38: note: in expansion of macro 'PROGMEM'
 const char S_JSON_DISPLAYINTERFACE[] PROGMEM = "display";
                                      ^
In file included from /Users/rileynichols/Downloads/esp8266_deauther-2.5-2.0-rc2/esp8266_deauther/esp8266_deauther.ino:14:0:
language.h:631:12: error: 'const char S_JSON_DISPLAYINTERFACE [8]' previously defined here
 const char S_JSON_DISPLAYINTERFACE[] PROGMEM = "display";
            ^
In file included from /Users/rileynichols/Library/Arduino15/packages/deauther/hardware/esp8266/2.7.2/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6:0,
                 from /Users/rileynichols/Library/Arduino15/packages/deauther/hardware/esp8266/2.7.2/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,
                 from /Users/rileynichols/Library/Arduino15/packages/deauther/hardware/esp8266/2.7.2/cores/esp8266/Arduino.h:32,
                 from sketch/esp8266_deauther.ino.cpp:1:
/Users/rileynichols/Library/Arduino15/packages/deauther/hardware/esp8266/2.7.2/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:25:130: error: redefinition of 'const char S_JSON_DISPLAY_TIMEOUT []'
   #define PROGMEM __attribute__((section( "\".irom.text." __FILE__ "." __STRINGIZE(__LINE__) "."  __STRINGIZE(__COUNTER__) "\"")))
                                                                                                                                  ^
sketch/settings.h:82:37: note: in expansion of macro 'PROGMEM'
 const char S_JSON_DISPLAY_TIMEOUT[] PROGMEM = "displayTimeout";
                                     ^
In file included from /Users/rileynichols/Downloads/esp8266_deauther-2.5-2.0-rc2/esp8266_deauther/esp8266_deauther.ino:14:0:
language.h:632:12: error: 'const char S_JSON_DISPLAY_TIMEOUT [15]' previously defined here
 const char S_JSON_DISPLAY_TIMEOUT[] PROGMEM = "displayTimeout";
            ^
/Users/rileynichols/Downloads/esp8266_deauther-2.5-2.0-rc2/esp8266_deauther/esp8266_deauther.ino: In function 'void setup()':
esp8266_deauther:91:5: error: 'settings' is not a class, namespace, or enumeration
     settings::load();
     ^
esp8266_deauther:103:9: error: 'settings' is not a class, namespace, or enumeration
     if (settings::getDisplaySettings().enabled) {
         ^
esp8266_deauther:120:20: error: 'settings' is not a class, namespace, or enumeration
     setWifiChannel(settings::getWifiSettings().channel);
                    ^
esp8266_deauther:123:9: error: 'settings' is not a class, namespace, or enumeration
     if (settings::getCLISettings().enabled) {
         ^
esp8266_deauther:132:9: error: 'settings' is not a class, namespace, or enumeration
     if (settings::getWebSettings().enabled) wifi::startAP();
         ^
/Users/rileynichols/Downloads/esp8266_deauther-2.5-2.0-rc2/esp8266_deauther/esp8266_deauther.ino: In function 'void loop()':
esp8266_deauther:156:9: error: 'settings' is not a class, namespace, or enumeration
     if (settings::getAutosaveSettings().enabled
         ^
esp8266_deauther:157:42: error: 'settings' is not a class, namespace, or enumeration
         && (currentTime - autosaveTime > settings::getAutosaveSettings().time)) {
                                          ^
esp8266_deauther:161:9: error: 'settings' is not a class, namespace, or enumeration
         settings::save(false);
         ^
exit status 1
'const char S_JSON_VERSION [8]' previously defined here`

I have ArduinoJSON version 5.10.0 I have tried installing 5.0 5.1 5.2 5.3 5.4 And I always get the error Please install ArduinoJSON version 5....

AleksZavg, 2020-10-08:
Install Arduino JSON 5.1.0 version, and delete this: C:\Users\AleksZavg\Documents\Arduino\
Try install with guide https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide

Thman558, 2020-10-08:
Got it working now! Thanks

closed 2020-10-08 by Thman558