SpacehuhnTech/esp8266_deauther #1330: wrong OLED pins mentioned

VIS-WA, 2020-10-18:
There's a mistake in declaration of pins for OLED in the example section 

**Location**
(Wiki)
In '[Example setup with I2C OLED](https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Setup-Display-&-Buttons#example-setup-with-i2c-oled)' in [Setup Display & Buttons](https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Setup-Display-&-Buttons#setup-display--buttons).

**Additional context**
Mine didnt work with pins mentioned for SDA, SCL for I2C OLED given in the example connections. So i referred to other sources and found out that the pins mentioned are swapped. i.e SDA should be connected to D2 and SCL to D1 of nodeMCU

Therealjosephchrzempiec, 2021-01-10:
Hello did you get your oled to work? I have been trying to get mine working nd so far no luck?

joseph

Chesterphillip, 2021-01-10:
I need to research it more . I’m probably not doing it right

spacehuhn, 2021-01-10:
Swapping SDA and SCL is a good idea to test if nothing works. I've also encountered some that had SDA and SCL labels swapped.  
As explained in [Setup Display & Buttons](https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Setup-Display-&-Buttons#setup-display--buttons), you can use any pins for a I2C connection. The ones mentioned there are just part of an **example** setup. The same pins are also used by the deauther boards from DSTIKE.

closed 2021-01-10 by spacehuhn

Therealjosephchrzempiec, 2021-01-11:
Hello I'm using the watch it looks like the first version of the watch. I'm trying to reporgram it. But the lack of coding skills are my problem. What library is the Oled using? Also I'm trying to use adafruit neo pixel library for the RGB LEd and it looks like it is using GPIO 9 (SD2) I don't know how to map a SD2 i have tried GPIO9 and i get a error. Sorry my skils are in hardware not much of software. I'm triyng.

Joseph

andrewbpowers, 2021-06-08:
> ...SDA should be connected to D2 and SCL to D1 of nodeMCU

Absolutely correct!
Normally SDA is connected to D2 and SCL to D1.
With a different wiring, I have not yet been able to put an OLED into operation.