SpacehuhnTech/esp8266_deauther #1335: Hỗ trợ cho ép07 không ạ

tienanh996tn, 2020-10-25:
> Please search for existing (open and closed) issues first to avoid duplicates.  
Also have a look at the [Wiki](https://github.com/spacehuhntech/esp8266_deauther/wiki).

tienanh996tn, 2020-10-25:
Mình có một con esp07 mình chạy code 2.0 thì không hỗ trợ cho esp07 admin giúp mình một bản được không

spacehuhn, 2020-10-26:
ESP-07 is a ESP8266 module with 1MB, so it is compatible to this project and you can use the 1MB bin file under releases.

tienanh996tn, 2020-10-26:
I tried it and it doesn't work

Được gửi từ iPhone của tôi

> Vào ngày 26 thg 10, 2020, lúc 18:12, Stefan Kremser <notifications@github.com> đã viết:
> 
> ESP-07 is a ESP8266 module with 1MB, so it is compatible to this project and you can use the 1MB bin file under releases.

tienanh996tn, 2020-10-26:
> ESP-07 là một mô-đun ESP8266 với 1MB, vì vậy nó tương thích với dự án này và bạn có thể sử dụng tệp bin 1MB trong các bản phát hành.

Mình đã tải với 1mb mà nó không có hoạt động admin có thể chỉ chi tiết cách để cho con esp-07 hoạt động ở phiên bản 2.0 được không

spacehuhn, 2020-10-26:
Sorry but I don't understand your problem.

tienanh996tn, 2020-10-26:
> Xin lỗi nhưng tôi không hiểu vấn đề của bạn.

Hướng dẫn mình cài cài đặt trên esp-07 được không mình thử vài cách rồi nhưng toàn thất bại

Bommlon, 2020-10-26:
Please translate your question to English so we are able to help you.

tienanh996tn, 2020-10-26:
> Vui lòng dịch câu hỏi của bạn sang tiếng Anh để chúng tôi có thể giúp bạn.

I have a problem about my esp-07 loading version 1.6, it works but upgrading to 2.0 or 2.1 won't work anymore admin help me to install, can I download the 1mb version but can't  admin makes a video on how to install esp-07

tienanh996tn, 2020-10-27:
> Xin lỗi nhưng tôi không hiểu vấn đề của bạn.

I have a problem with the esp-07 version that is downloading version 1.6, it works, but when I upgrade to 2.0 or 2.1, I can't anymore admin help me how to install it, can I download the 1mb version?  install is not esp-07

closed 2021-02-01 by stale[bot]