SpacehuhnTech/esp8266_deauther #1338: Cannot save settings, stations and after reload, AP does not appear

TanimSk, 2020-10-27:
I am using ESP-01 and have flashed "deauther_2.1.0_1mb.bin" in it. Everything worked as expected but **after scanning Access Point and reloading it, the APs don't appear** (though my home router is of 2.4GHz).
Also, **I can't save anything like settings(in fact there's nothing in settings), APs, customized SSIDs for Beacon attack**.
The beacon attack is running well, but I can't perform "Deauth" attack as the APs don't appear.

NOTE: I have flashed the exact same binary into my NodeMCU, everything works fine, but I don't know why my esp01 is behaving like this.

Environment:
 - OS: Windows 7
 - Browser: Chrome
 - Version: esp8266_deauther-2.1.0
 - Hardware: ESP-01
 - Flash Tool: ESP_flash_download_tool_3.8.5
[image]
[image]

tienanh996tn, 2020-10-28:
> Tôi đang sử dụng ESP-01 và đã nhấp nháy "deauther_2.1.0_1mb.bin" trong đó. Mọi thứ hoạt động như mong đợi nhưng **sau khi quét Access Point và tải lại nó, các AP không xuất hiện** (mặc dù bộ định tuyến nhà tôi là 2.4GHz). 
> Ngoài ra, **tôi không thể lưu bất kỳ thứ gì như cài đặt (thực tế là không có gì trong cài đặt), AP, SSID tùy chỉnh cho cuộc tấn công Beacon** . 
> Cuộc tấn công bằng đèn hiệu đang chạy tốt nhưng tôi không thể thực hiện cuộc tấn công "Deauth" vì các AP không xuất hiện.
> 
> LƯU Ý: Tôi đã flash cùng một tệp nhị phân chính xác vào NodeMCU của mình, mọi thứ đều hoạt động tốt, nhưng tôi không biết tại sao esp01 của tôi lại hoạt động như thế này.
> 
> Môi trường:
> 
> * Hệ điều hành: Windows 7
> * Trình duyệt: Chrome
> * Phiên bản: esp8266_deauther-2.1.0
> * Phần cứng: ESP-01
> * Công cụ Flash: ESP_flash_download_tool_3.8.5
> 
> [image]
> [image]

Xin cách cài đặt

TanimSk, 2020-10-29:
@spacehuhn ### Please help!

closed 2020-12-04 by TanimSk