SpacehuhnTech/esp8266_deauther #1345: pwned Network not showing up

Nipeno, 2020-11-07:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

**Describe the error**
A clear and concise description of what the problem is.
What do you think causes it?

I just got the Deauther OLED V6, worked fine earlier, then I changed the name of the network from pwned to something different, now it just doesn't show up on my Wi-Fi. I already flashed the software, still no luck, please help.

**Expected behavior**
Wifi network should show up

**Environment (please complete the following information):**
 - OS: iOS
 - Browser Safari
 - Hardware DSTIKE
 - Deauther Version 2.5.0

**Attempts**
Flashing, using another phone to try to connect (didn't show up)

oldrox, 2020-11-08:
The problem is that you have hidden the web interface, not the SSID.
I had this problem. I solved it with the latest firmware via esptool. Do not forget to hold down the flash button on the board before starting the firmware process.

sircarti, 2020-11-25:
> Yes
>
> **Describe the error**
> A clear and concise description of what the problem is.
> What do you think causes it?
>
> I just got the Deauther OLED V6, worked fine earlier, then I changed the name of the [...]

Nothing realated to that my bad. But I also have a DSTIKE running that version does the attack function work for you? For me it hasn’t really. kind of a bummer.

spacehuhn, 2021-01-14:
You can try resetting the settings: https://github.com/SpacehuhnTech/esp8266_deauther/wiki/FAQ#how-to-do-a-reset

closed 2021-04-22 by stale[bot]