SpacehuhnTech/esp8266_deauther #1347: problem with html editing

garustea, 2020-11-13:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes/No

**Describe the error**
A clear and concise description of what the problem is.
What do you think causes it?

```
Error/Compile/Output Log
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 1.0]
 - Hardware [e.g. DSTIKE, DIY]

**Attempts**
What have you already tried and didn't work out.

**Additional context**
Provide as much information as possible, better too much than too little!
If you don't use this template, your issue might be closed and tagged invalid!

Bommlon, 2020-11-23:
Please add more information to make it easier for us to help you out.

SystemStack, 2021-01-08:
When you added Brazilian Portuguese you did everything almost perfect, but you instantiated the variable with a hyphen in the name, which is not allowed. To fix this, you'll want to change 
` const char pt-brlang[] PROGMEM = {…}; `
to 
` const char ptbrlang[] PROGMEM = {…}; `

then update line 65 of webfiles.h to use that variable:
`if(!SPIFFS.exists(String(F("/web/lang/pt-br.lang.gz"))) || force) progmemToSpiffs(ptbrlang, sizeof(ptbrlang), String(F("/web/lang/pt-br.lang.gz")));`

closed 2021-04-16 by stale[bot]