SpacehuhnTech/esp8266_deauther #1352: The attack effect is not good, and a web config bug

People-11, 2020-11-16:
I am using OLED V6 and running V2.5 RC2 on it, the issues I met are as follows.
1. When I start Beacon attack, there are only a few SSIDs are broadcast, even no SSID. Enableling Beacon interval can not solve the problem.
2. The Beacon interval option in the web is oppisite. That means when you check it, it is actually turned off, and when you check it and uncheck it again, it is actually turned on.
I bought OLED V6 from Taobao and found this issue, but bcs of I have it over 7 days so I can not get the refund. So PLZ solve this issue then make it work.

sircarti, 2020-11-25:
same. I have a DSTIKE v2 watch. And the deauth function just doesn’t seem to work. im not sure why im new to all this.

Bommlon, 2020-12-09:
@sircarti have a look at the [Wiki](https://spacehuhn.wiki/deauther/troubleshooting/).

closed 2021-03-16 by stale[bot]