SpacehuhnTech/esp8266_deauther #1353: SSD1306 display problem and arduino not working problem after compiling

LuckyFishGeek, 2020-11-17:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

I have this board https://www.banggood.com/Wemos-D1-Esp-Wroom-02-Motherboard-ESP8266-Mini-WiFi-NodeMCU-Module-p-1224577.html 

**Describe the error**
A clear and concise description of what the problem is.
What do you think causes it?

I directly flashed the firmware esp8266_deauther_2.5.0-rc2_DSTIKE_DEAUTHER_OLED_V3_5.bin and it works normally

But after compiling with arduino, it does not work
I read the instructions on the wiki, modified the content of the A_config.h file, and uncommented #define DEAUTHER_OLED_V3_5

SSD1306 monitor does not display after arduino compilationError

**To Reproduce**
[image]
[image]
[image]

After flashing the firmware, the screen does not display, and the PWNED hotspot cannot be connected

**Environment (please complete the following information):**
 - OS: windows7 64bit
 - Browser   chrome
 - Version  85.0.4183.121
 - Hardware https://www.banggood.com/Wemos-D1-Esp-Wroom-02-Motherboard-ESP8266-Mini-WiFi-NodeMCU-Module-p-1224577.html 

**Attempts**
 already tried and didn't work out.

Thanks friends for your help

d1qqn, 2020-11-22:
i have the same board and my problem is, that the display is not clear.

[image]

does anyone know what to do here?

LuckyFishGeek, 2020-11-22:
@d1qqn   Brother, how do you deal with the display problem?

fish-hue, 2020-11-22:
I have two deauthors, different versions, and if I could get my money back I would (sorry to my old deauthors). They work, but reset from time to time.

For $20 and a SD card, consider making your own pwnagotchi, more useful in my opinion and more for your money.

If you get a screen be sure to get a board with a header already soldered.

*My apologies as this is not related directly to your question, but if you have a chance I suggest going to the pwnagotchi.ai website and you can see what I am talking about.

RChadwick7, 2020-11-22:
It looks like you have a cracked or damaged display. All the connections come in the bottom, and if a corner is cracked (Very easy to do), you can get the effect where every other line is off. The solution is to solder a new display on the PCB.

d1qqn, 2020-11-22:
> @LuckyFishGeek

I use the web interface 

> @fish-hue 

thanks for the recommendation, but I already have a pwnagotchi and I'm very happy with this little toy ...
i can recommend it, too ;)

> @RChadwick7

It seems you are right. .. My display has broken off a bit in the lower right corner... 

But I've already read on the Internet that someone had the same problem.
his solution was to comment out
#define SSD1306_128_64
instead of 
#define SSD1306_128_32
in the "Adafruit_SSD1306.h" file

But this file doesn't exist in this project, right?

closed 2020-11-25 by LuckyFishGeek