SpacehuhnTech/esp8266_deauther #1355: No internet connectiom

Ritesh8732, 2020-11-27:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

**Describe the error**
A clear and concise description of what the problem is.
What do you think causes it?

I connected my laptop to the pwned successfully but when opened http://192.168.4.1/  its showing no "internet connection" . In short the  webpage is not opening.

**Environment (please complete the following information):**
 - OS: windows 10
 - Browser : Brave
 - Version [e.g. 1.0]
 - Hardware : lolin Nodemcu v3

Alpha-1729, 2020-12-23:
Same issue for me also.

Ritesh8732, 2020-12-23:
Hello
Still I haven't able to find the solution for that issue , end up using the
first method using  .Bin  file.

Thanks