SpacehuhnTech/esp8266_deauther #1356: Web interface error 192.168.4.1

mohibkhan011, 2020-11-28:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

**Describe the error**
A clear and concise description of what the problem is.
What do you think causes it?

I've just received the device and I start connecting to web interface 192.168.4.1 but its not going to the dashboard. 

Error/Compile/Output Log
Error 404 file not found

**Screenshots**
[image]

**Environment (please complete the following information):**
 - OS: [ Android]
 - Browser [chrome, ]
 - Version [e.g. 2.5.0]
 - Hardware [ DSTIKE]

Alpha-1729, 2020-12-23:
Same error for me also

Bommlon, 2021-01-07:
Did you make sure you're connecting to the correct URL?

Alpha-1729, 2021-01-07:
Yes

spacehuhn, 2021-01-12:
On some phones, you must disable the mobile data connection to be able to access `192.168.4.1`.

Nomade12, 2021-01-15:
Having the same issue... Flashed with the current code and everything works in serial mode. Web interface is completely missing..
I tried to connect both to 192.168.4.1 or deauth.me with both Mozilla desktop browser, Chrome on android phone and Safari on iPad
I'm pasting here my sysinfo and a screen below.

23:24:19.457 -> # sysinfo
23:24:19.457 -> [======== SYSTEM INFO ========]
23:24:19.457 -> RAM usage: 47896 bytes used [59%], 34024 bytes free [41%], 81920 bytes in total
23:24:19.491 -> 
23:24:19.491 -> Current WiFi channel: 1
23:24:19.491 -> AP MAC address: da:bf:c0:14:b0:a4
23:24:19.491 -> Station MAC address: 00:10:66:41:0e:a7
23:24:19.491 -> SPIFFS: 40960 bytes used [1%], 3080192 bytes free [98%], 3121152 bytes in total
23:24:19.491 ->         block size 8192 bytes, page size 256 bytes
23:24:19.491 -> Files: 
23:24:19.491 ->   autostart.txt 19 bytes
23:24:19.491 ->   names.json 2 bytes
23:24:19.525 ->   scan.json 785 bytes
23:24:19.525 ->   settings.json 507 bytes
23:24:19.525 ->   ssids.json 318 bytes
23:24:19.525 -> [WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: true, captive-portal: false
23:24:19.525 -> ===============================

[image]

Nomade12, 2021-01-16:
OK, i rolled back and uploaded v.2.1.4 on my nodemcu, this is the versione before you replaced SPIFFS with LittleFS and it worked beautifully.
Now my web interface gets correctly viewed.
I dont know how much this can help in resolving this bug, but i hope it gets resolved soon.

spacehuhn, 2021-01-30:
Thank you for reporting. I pushed a few changes recently that should resolve the problem.