SpacehuhnTech/esp8266_deauther #1359: Esp32 Compatible

riocamcarot, 2020-12-09:
> Please search for existing (open and closed) issues first to avoid duplicates.  
Also have a look at the [Wiki](https://github.com/spacehuhntech/esp8266_deauther/wiki).  

Hi, I'm really interested in this project and I am having really fun with it. I'm just wondering if the deauther is compatible with ESP32 because I'm going to buy another Esp32 for my experiments.
Thanks a lot for sharing your project and I hope you will achieve more success on what you're doing!

riocamcarot, 2020-12-09:
I've found the answer. Sorry for bothering!!

closed 2020-12-09 by riocamcarot

spacehuhn, 2021-12-12:
Hi! Since there were so many issues on this repo about supporting the ESP32, I published a blog post to bring some clarity to this topic: https://blog.spacehuhn.com/esp32-deauther/ 
TL;DR: It's made for the ESP8266, and I have no plans to port this application to the ESP32. But it's open-source, so you're free to do it yourself if you want.