SpacehuhnTech/esp8266_deauther #1361: Problem with the code "Dstike Deauther Watch V2"

ScorpinS, 2020-12-13:
Hi, it turns out that I bought this watch, download the code for my watch and open it with the arduino IDE, by uncommenting the corresponding board for me in configuration ...

**A_config.h**
```
/* ---------------------------------------------------------- */
/* ===>> !!! Uncomment the board that you're using: !!! <<=== */
/* ---------------------------------------------------------- */
// #define DSTIKE_DEAUTHER_WRISTBAND
// #define DSTIKE_DEAUTHER_WATCH
 #define DSTIKE_DEAUTHER_WATCH_V2
// #define DSTIKE_DEAUTHER_MINI
// #define DSTIKE_DEAUTHER_MINI_EVO
```

I get a compilation error

```
In file included from sketch\CLI.h:16:0,
                 from C:\Users\Kimory\Desktop\esp8266_deauther-2.5.0-rc2\esp8266_deauther\esp8266_deauther.ino:33:
DisplayUI.h:27:20: fatal error: DS3231.h: No such file or directory
 #include <DS3231.h>
                    ^
compilation terminated.
exit status 1
DS3231.h: No such file or directory
```

**esp8266_deauther.ino**
```
32#include "Attack.h"
33#include "CLI.h"
34#include "DisplayUI.h"
```

**CLI.h**
```
15#include "Attack.h"
16#include "DisplayUI.h"
17#include "led.h"
```

**DisplayUI.h**
```
26#ifdef RTC_DS3231
27#include <DS3231.h>
28#endif
```

What I can do?
(If I comment the line, I get many more errors)
Beforehand thank you very much

timothysharp, 2020-12-29:
I get the same issue and I can’t find a fix

Bommlon, 2021-01-07:
Which firmware version did you use?

ScorpinS, 2021-01-07:
@Bommlon  esp8266_deauther-2.5.0-rc2

timothysharp, 2021-01-07:
Yea I used 2.5.0 rc2 and any other software doesn’t function correctly also but if fails to send packets like if I deauthed a wap it would send like 21/20 and fail to work

ScorpinS, 2021-01-07:
@timothysharp  too bad, we hope they solve it soon :3

spacehuhn, 2021-01-11:
Hello everyone, just made some changes to make the whole process easier.
Follow the updated instructions here: https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide

Make sure to have the latest code and latest deauther board package installed.
Step 5 and 6 are the important ones. In your case, select `DSTIKE` as board and `DSTIKE Deauther Watch V2` as config.

ScorpinS, 2021-01-11:
@spacehuhn ok entiendo, lo probare
Muchas Gracias

closed 2021-04-19 by stale[bot]