SpacehuhnTech/esp8266_deauther #1367: Change network name and password (security request)

sirakai-pc, 2020-12-26:
I want to be able to change the network name or at least the password for security purposes.. If there's the chance that someone knows this project he surely would know the password(Google would be is friend).. The more common reason is wordlist cracking.
The ability and a clear documentation of the feature.

sirakai-pc, 2020-12-27:
I've already seen [davedarko's work](https://gist.github.com/davedarko/87183b00e77ffb8fc59f89bf3b23d561) but you shoudn't definitely have your laptop or pc around you to demonstrate the vulnerability of these wifi networks, as I understand I can only change the wifi credentials by running the arduino code by a pc or a rooted android(if they're nearby the deauther)

HA4ever37, 2021-01-01:
This feature is implemented already. Follow these steps:
- Connect to "pwend" network
- Open web browser and enter this URL "http://192.168.4.1/settings.html"
- Change the network name and password
- Click on the "save" button and restart the ESP8266

closed 2021-01-01 by sirakai-pc