SpacehuhnTech/esp8266_deauther #1374: password

aaronreilly17, 2021-01-08:
> Please search for existing (open and closed) issues first to avoid duplicates.  
Also have a look at the [Wiki](https://github.com/spacehuhntech/esp8266_deauther/wiki).  
what is the pass word?

marco-porto, 2021-01-09:
Hello @aaronreilly17, password for the wifi hotspot is `deauther`😀

aaronreilly17, 2021-01-09:
> Hello @aaronreilly17, password for the wifi hotspot is `deauther`😀

hi mate it doesnt work for me keeps saying unable to connect u know what i could try do to fix?

spacehuhn, 2021-01-11:
You can try a settings reset [How to do a reset](https://github.com/SpacehuhnTech/esp8266_deauther/wiki/FAQ#how-to-do-a-reset) or (re)install the latest firmware [Installation](https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Installation)

closed 2021-04-19 by stale[bot]