SpacehuhnTech/esp8266_deauther #1375: Everthing.

Therealjosephchrzempiec, 2021-01-09:
> Please search for existing (open and closed) issues first to avoid duplicates.  
Also have a look at the [Wiki](https://github.com/spacehuhntech/esp8266_deauther/wiki).  

Hello i have checked most closed errors and i did find a few that i have tired and go no luck with it.  My watch looks like the first version maybe i got it from banggood a few months back. So my question is the oled screen is it SPI or I2C? Also i have tired adafruit oled library because i do have another esp board and i2c screen  and that works but i can not get this one to work.

I have tried this library [https://github.com/ThingPulse/esp8266-oled-ssd1306](url) same thing i can not get the screen to work.

Second question is the buttons and neopixel. Is this the setup for them correctly [https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Setup-Display-&-Buttons](url) ?

Joseph

Therealjosephchrzempiec, 2021-01-10:
Hello can someone tell me if this is a I2c Oled or a Spi oled, also what are the pins for the buttons and menu rocker switch? 
I tried to look at the site link [https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Setup-Display-&-Buttons](url) I have looked and tried to figure out what pins are what. I'm lost can someone please help me?

Joseph

spacehuhn, 2021-01-11:
Hello Joseph.
I've pushed some updates today to make this process easier.
1. Download the latest code and install the latest deauther boards in Arduino IDE (for more info see [compiling-using-arduino-ide](https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide). 
2. In Arduino you can now go to `Tools` > `Board`  and select `DSTIKE Deauther`.
3. Then at `Tools` > `Deauther Config` select `DSTIKE Deauther Watch`.
4. To make sure the changes take affect and no old settings mess this up, go to `Tools` > `Erase Flash` and select `All Flash Contents`
5. Select the right port (`Tools` > `Port`) and click upload.

closed 2021-01-11 by spacehuhn

Therealjosephchrzempiec, 2021-01-11:
Hello is there a json link for this? Because i couldn't find the boaard in he board manger.

Joseph

spacehuhn, 2021-01-11:
Yes, just follow these installation instructions: https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide

Therealjosephchrzempiec, 2021-01-11:
Hello i installed everything board manger i selected it the watch in the tools and the config with no problem. Everything upload with no problem. However I'm not getting nothing on the Screen. Not sure what to do next. I installed everything from the board manger correctly.

Joseph

spacehuhn, 2021-01-12:
Then I'm afraid, it's probably a hardware fault.

Therealjosephchrzempiec, 2021-01-12:
I can load the 2.5.0 bin file with no problem and the Oled and everything including the buttons work with no problem.

Joseph