SpacehuhnTech/esp8266_deauther #1379: Watchdog resets

Laurent-Jongbeys, 2021-01-16:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes/No

**Describe the error**
On a ESP8266EX NodeMCU, the script keeps getting reseted.
On the third cycle appears an invalid hash error on loading the settings.
Furthermore, cant see nothing on my SSD1306_I2C display.
V1.6 works fine with same hardwares (testing on two slightly different boards)
Error/Compile/Output Log
ets Jan  8 2013,rst cause:4, boot mode:(3,4)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

Mounting SPIFFS...OK
Loading settings...OK
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: false
STARTED! \o/
2.6.0

 ets Jan  8 2013,rst cause:4, boot mode:(3,4)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

Mounting SPIFFS...OK
Loading settings...OK
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: false
STARTED! \o/
2.6.0

 ets Jan  8 2013,rst cause:4, boot mode:(3,4)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

Mounting SPIFFS...OK
Formatting SPIFFS...OK
Formatting EEPROM...OK
Loading settings...Invalid Hash
Settings reset to default
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: false
STARTED! \o/
2.6.0

**Environment (please complete the following information):**
 - OS: OsX 
 - Version OSX 10.15, Arduino IDE 1.8.13
 - Hardware DIY/Node MCU

**Attempts**
Reset script, downgrading board definitions
Back to latest definitions.

Laurent-Jongbeys, 2021-01-17:
UPDATE: Using the provided node mcu board definitions, selecting the Deauther config option "none" in the board config dropdown, the script runs well, the web interface is accessible and so forth.
Can't find a way to enable the SSD1306_I2C display without the script going into the reset cycle though. Tried different pins with no results. It runs so smoothly with V.16, i cant figure out why.

Laurent-Jongbeys, 2021-01-17:
LAST UPDATE: Project is working. In order to use the SSD1306_I2C oled, i had to manually enable "#define NODEMCU" in the A_config.h in conjunction with selecting "Display_expample_i2C" in the Deauth config dropdown from the board config in arduino IDE.
It was getting on my nerves.

closed 2021-01-17 by Laurent-Jongbeys

spytaspund, 2022-10-16:
Exactly the same situation. Only I did as written in the solution and NodeMCU still reboots every 3-7 seconds. Any guesses?
SERIAL OUTPUT:
` ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

Mounting SPIFFS...OK
Loading settings...OK
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: false
STARTED! \o/
2.6.1

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

Mounting SPIFFS...OK
Loading settings...OK
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: false
STARTED! \o/
2.6.1
`