SpacehuhnTech/esp8266_deauther #1383: Screen Clock Doesn't Update

zurpv1, 2021-01-20:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

**Describe the error**
I just installed v2 of the DSTIKE Watch v2 deauther software. I tab over the Clock screen. I adjust the time. The time doesn't refresh itself. If I turn the watch off and then back on, the time does update.

```
Error/Compile/Output Log 
There are no errors received.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Clock'
2. Adjust time
3. Wait 
4. Time doesn't update/refresh
5. Leaving the Clock screen and returning doesn't correct the issue either

**Expected behavior**
I expect the time to refresh as a clock should

**Screenshots**
None available

**Environment (please complete the following information):**
 - OS: Spacehuhn 2.6.0
 - Browser NA
 - Version 2.6.0
 - Hardware DSTIKE Watch v2

**Attempts**
I am still learning the code. I am not very knowledgeable yet.

**Additional context**
Please let me know if you need any other information. Thanks.

Nikakz, 2021-01-20:
Do you have define RTC_DS3231 enabled? Take a look at the RTC settings in the code. Good luck

zurpv1, 2021-01-20:
Thanks for the tip, Nikakz. I downloaded the entire package and left the src structure intact. I would think that I would encounter errors during verification. Also, it does keep the time, it just doesn't refresh as it should. I am thinking it has something to do with the display refresh. I will try to check on your suggestion. Take care and I will update this with any helpful information.

Nikakz, 2021-01-20:
I'm having trouble getting the led to work, maybe you can help me out, read my post. Thanks

zurpv1, 2021-01-22:
I have reached the limit of my knowledge on this. I believe this is a bug and will submit it.

closed 2021-04-30 by stale[bot]