SpacehuhnTech/esp8266_deauther #1385: Clock Doesn't Refresh

zurpv1, 2021-01-22:
**Describe the bug**
The clock doesn't refresh once the watch is on. The only way it refreshes is when the watch is turned off and back on or when flashed with firmware.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn watch on
2. Click on 'Clock'
3. Wait
4. Doesn't refresh

**Expected behavior**
I expect the time to refresh
**Screenshots**
NA

**Environment (please complete the following information):**
 - OS: 2.6.0
 - DSTIKE Watch v2
 - Version 2.6.0
 - Hardware DSTIKE

**Additional context**
I have tried editing the code for the past 3 days without any success. 
As stated I am unable to get the clock to refresh at all. I tried to edit the drawClock()

void DisplayUI::drawClock() {
 
    String clockTime = String(clockHour);
    clockTime += ':';
    if (clockMinute < 10) clockTime += '0';     
    clockTime += String(clockMinute);
    display.drawString(64, 20, clockTime);     
}

I created a separate function and added clockSeconds to this new function. I tried adding a loop and display.clear() with no success. I think I am hitting a wall because I don't understand the nesting of the voids.  

// <<---------------------------- Added 1-22-2021 ------------------------>> \\

void DisplayUI::drawClockDisplay() {

    String clockTime = String(clockHour);
    clockTime += ':';
    if (clockMinute < 10) clockTime += '0';     
    clockTime += String(clockMinute);
    clockTime += ':';
    if (clockSecond < 10) clockTime += '0';     
    clockTime += String(clockSecond);
    display.drawString(61, 20, clockTime); 
}  

// <<----------------------------------------------------------------------->> \\

I did have success in creating a Clock sub menu and divided the original Clock to: Clock Display and Clock Time Set. This works better for me since I don't have to worry about accidentally changing the time when viewing the Clock Display option.

//  <<----------------------------- Added 1-21-2021 ------------------------------>> \\
    
    // CLOCK MENU
    createMenu(&clockMenu, &mainMenu, [this]() {
        addMenuNode(&clockMenu, D_CLOCK_DISPLAY, [this]() { // CLOCK 
            mode = DISPLAY_MODE::CLOCK_DISPLAY;
            display.setFont(ArialMT_Plain_24);
            display.setTextAlignment(TEXT_ALIGN_CENTER);           
        });
        addMenuNode(&clockMenu, D_CLOCK_SET, [this]() { // CLOCK SET TIME
            mode = DISPLAY_MODE::CLOCK;
            display.setFont(ArialMT_Plain_24);
            display.setTextAlignment(TEXT_ALIGN_CENTER);
        });
    });

//  <<------------------------------------------------------------------------------->> \\

Nikakz, 2021-01-22:
Hello, complicated these codes when we are starting. I solved my problem with Led Neopixel, but now I face problems with the screen where there are boring strokes under the word scan in the menu and I can't solve it, it happened after I had to invert the screen, because it was wrong here for me. I don't know how to solve this. I hope you can solve your problem.

Nikakz, 2021-01-23:
You saw in DisplayUi.cpp:
#ifdef RTC_DS3231
     bool h12;
     bool PM_time;
     clock.setClockMode (true);
     clockHour = clock.getHour (h12, PM_time);
     clockMinute = clock.getMinute ();
#else
     clockHour = random (12);
     clockMinute = random (60);
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
Where clock.setClockMode (false) for put (true)??? in my configuration say false and I put true

zurpv1, 2021-01-23:
> You saw in DisplayUi.cpp:
> #ifdef RTC_DS3231
> bool h12;
> bool PM_time;
> clock.setClockMode (true);
> clockHour = clock.getHour (h12, PM_time);
> clockMinute = clock.getMinute ();
> #else
> clockHour = random [...]

Yes, I saw that. I had tried using it initially. I believe the reason the clock doesn't refresh is because the program is set to only .update at startup, reset, scan, and button press. I came to this conclusion by looking at void loop in the .ino file. My best guess at this time. 

I am hoping one of the developers fixes this soon. I am sure this is an easy fix if you know what you are doing. lol.

spacehuhn, 2021-01-23:
Thank you for your help. Just pushed some changes based on your reports and suggestions.

zurpv1, 2021-01-23:
> Thank you for your help. Just pushed some changes based on your reports and suggestions.

Spacehuhn, thank you for all of your hard work. Can't wait to try it out! Take care.

zurpv1, 2021-01-25:
The clock is working great. Thank you!