SpacehuhnTech/esp8266_deauther #1392: Attack stops after a while dosen´t work

Petros626, 2021-01-29:
Hallo Leute,

spacehuhn hat bei den FAQ gesagt man soll "attackTimeout" auf 0 setzen, damit die Attacke durchgängig läuft.

Meine Frage wäre jetzt, ob das in diesem Code Abschnitt (Attack.cpp) erfolgen muss ? Habe es mal auf 0 gesetzt und gemerkt, dass gar keine Attacke mehr stattfindet (Bedingung nie true immer false):

// each second
    if (currentTime - attackTime > 1000) {
        attackTime = currentTime; // update time
        updateCounter();

        if (output) status();     // status update
        getRandomMac(mac);        // generate new random mac
    }

Vielen Dank für eure Rückmeldungen

spacehuhn, 2021-01-30:
You don't have to edit the code. There should be a `attacktimeout` setting at the settings page for this (192.168.4.1/settings.html).

Petros626, 2021-01-30:
i mean the code on the hardware stops after a few minutes. have the deauther mini and oled v5, as soon as the attack starts it stops after the display goes off.

i changed a condition in the code to work around this problem. just didn't know what is meant by attackTimeout as i don't use the web application

geo777535, 2021-03-23:
> You don't have to edit the code. There should be a `attacktimeout` setting at the settings page for this (192.168.4.1/settings.html).

after installing firmware 2.6 on esp8266 (nodemcu), when I choose in settings to hide ssd, earlier on a version below 2.1.0 on Windows it did not show the "hidden network" 
[image]
now it shows like this. I don’t want to show "hidden networks" an inscription and unmasked the module esp8266 (nodemcu).
second problem: after the settings, the LED is turned off, after the esp8266 (nodemcu) module is turned on, the blue LED is on.

closed 2021-06-28 by stale[bot]