SpacehuhnTech/esp8266_deauther #1398: 2.6.0 A_config.h

DOKGMC, 2021-02-03:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

Cannot figure how to change password or ssid in the config file A_config.h at #define AP_SSID "pwned".  Using a Nodemcu esp8266. This was for 2.100. I hoped it work for 2.6.0. But the A_config.h files are stup different.

There are 2 #define AP_SSID "pwned" , along with  #define DEFAULT_SSID "pwned"
Same with changing the password. 2=#define AP_PASSWD "deauther"

I took this out of the 2.6.0 A_config.h
_____________________________________________
#ifndef DEFAULT_SSID
  #define DEFAULT_SSID "pwned"
#endif /* ifndef DEFAULT_SSID */

// ===== ACCESS POINT ===== //
#ifndef AP_SSID
  #define AP_SSID "pwned"
#endif /* ifndef AP_SSID */

  // ===== ACCESS POINT ===== //
 #define AP_SSID "pwned"<<<<<
 #define AP_PASSWD "deauther"
 #define AP_HIDDEN false
 #define AP_IP_ADDR {192, 168, 4, 1}
______________________________________________

#ifndef AP_PASSWD
  #define AP_PASSWD "deauther"
#endif /* ifndef AP_PASSWD *

   // ===== ACCESS POINT ===== //
 #define AP_SSID "pwned"
 #define AP_PASSWD "deauther"<<<<<
 #define AP_HIDDEN false
 #define AP_IP_ADDR {192, 168, 4, 1}

Where do i change these. Do i change all of them or just the the SSID and PASSWD in the  // ===== ACCESS POINT ===== //

*Attempts**
I tried changing them all. No go!
I tried changing in  // ===== ACCESS POINT ===== //  = No go also.

spacehuhn, 2021-02-03:
As far as I can see `DEFAULT_SSID` is unused, I'll remove it. `AP_SSID` however is used for resetting the settings back to default. If you can't see any change, try forcing a [settings reset](https://github.com/SpacehuhnTech/esp8266_deauther/wiki/FAQ#how-to-do-a-reset).

DOKGMC, 2021-02-03:
Ok, I removed the DEFAULT_SSID from  2.6.0 A_config.h, I also set 2x PASSED and SSID. in the  2.6.0 A_config.h. I used a new Node Ncu board as the other one took a crap. I also tried setting just the PASSWD and SSID in the // ===== ACCESS POINT ===== //.

I plugged in the Board uploaded a empty bin and then uploaded the edited 2.6.0 A_config.h, along with the 2.6.0 esp8266_deauther.ino. The default username and password never changed on both scenerios. I also tried a 4mb reset bin along with the edited A_config.h 2.6.0 esp8266_deauther.ino with the same results. 
Settings for nodemcu in Arduino .

Board: NodeMCU  
Upload Speed: 115200
Flash Size: 4MB "(FS:3MB OTA .~512k)"
Flash Erase:
Deather config: NodeMCU
Port: "COM12"

Here is my A_config.h settings.
[A_config.h](https://pastebin.pl/view/e68b0341)

closed 2021-05-12 by stale[bot]