SpacehuhnTech/esp8266_deauther #1400: Littlefs.h no such file or directory

iama10yearold, 2021-02-08:
Hello I need help with the deauther code because when ever I try to verify the code it keeps saying that little fs is not a file or directory please help me with this @spacehuhn

Tim-Hobbes, 2021-02-15:
First of all it is rude to ping someone in a issue when they are not needed to help.
Second if you open a issue it is very helpful for the person trying to help to include the steps you did/tutorial you followed. 

Looking at you issue I think you have not followed the installation/compilation instructions carefully enough. I suggest following this interactive guide: [deauth.me](http://deauth.me/). I found the link on the [installation Github Wiki](https://github.com/spacehuhntech/esp8266_deauther/wiki/Installation) page.

iama10yearold, 2021-02-15:
Sorry no need to be mean I just joined

closed 2021-05-28 by stale[bot]