SpacehuhnTech/esp8266_deauther #1403: button not working

wh1ter0z, 2021-02-14:
i flashed v2.6.0 nodemcu with nodemcu flasher and setup lcd with i2c 1306 

buttons didnt work ( d5 d6 d7) i conected gnd to d5 then not work

on v2.1.0 d5 d6 works but d7 (A) not work for me ...

how can i solve this problem ?

please help me !

unkn0wn-wifi, 2021-02-24:
Same here can we get some guidance here @spacehuhn ive tried multiple tries, and even when running the button test they all are on false when trying to click on them, this seems to me as a coding issue can you help?

im using a i2c SSD1306 display with a node mcu, i flashed it using the arduino flasher as i had to edit some code, i even tried replacing all of the 

"  #define BUTTON_UP 13
   #define BUTTON_DOWN 14
   #define BUTTON_A 12
"
Please can you help,

ps you may know me from kody's live stream when you joined and i spoke to you, UNKNOWN

wh1ter0z, 2021-02-28:
yes i can help you ...

but i cant join to kody's stream

give me a whatsapp or telegram or instagram id to help you 

my instagram ID is 4xtrom

MegaAndrey, 2021-03-07:
> yes i can help you ...
> 
> but i cant join to kody's stream
> 
> give me a whatsapp or telegram or instagram id to help you
> 
> my instagram ID is 4xtrom

I would be very happy if you can help me, I have a similar problem, my telegram @Ararelos

Nikakz, 2021-03-16:
[image]

Deniskyn, 2021-03-24:
how the scheme above affects the operation of the buttons?

FriendlyWizard23, 2021-03-29:
did you guys solve the problem?

wh1ter0z, 2021-04-16:
i solved this if you want to come in instagram : 4xtrom

Bommlon, 2021-04-18:
@wh1ter0z you know you could just post your solution right here, right?

strolll395, 2021-07-08:
@wh1ter0z your instagram doesn't exist anymore, would you mind just posting the fix directly?

wh1ter0z, 2021-07-09:
Sure, no problem.

#define BUTTON_UP 14 gpio 14 = d5 #define BUTTON_DOWN 12 gpio 12 = d6 #
define BUTTON_A 13

gpio0 = d3

---------------------
d1 d2 use for lcd

use left GND for buttons
@irh1m is my new IG
if u want more help tell me...

wh1ter0z, 2021-07-09:
Sure, no problem.

#define BUTTON_UP 14 gpio 14 = d5 #define BUTTON_DOWN 12 gpio 12 = d6 #
define BUTTON_A 0

gpio 0 = d3

---------------------
d1 d2 use for lcd

use left GND for buttons
@irh1m is my new IG
if u want more help tell me...

wh1ter0z, 2021-07-09:
check my   Repositories  <https://github.com/wh1ter0z?tab=repositories>

i uploaded edited project and changed something

strolll395, 2021-07-10:
Thank you for being so nice to make this fix easily accessible. It really helps and will save a ton of time in the future. I will try it out later and report back if it works

strolll395, 2021-07-10:
Uhm, sorry, I get the following error while compiling:
exit status 1
'USE_LED' was not declared in this scope

strolll395, 2021-07-10:
Nevermind, I got it. I removed the RGB LED part since I dont need it anyways.

wh1ter0z, 2021-07-10:
okay fine i will fix it

wh1ter0z, 2021-07-10:
sorry for errors i fixed now u can update A_config.h file

strolll395, 2021-07-10:
tested it and it works perfect now. thanks for taking your time and fixing it :)

wh1ter0z, 2021-07-10:
You're welcome!

ngocthuong234, 2021-08-17:
With only one esp8266, deauth + evil twin + Facebook, Google, Twitter,..
https://youtu.be/oKsRb4KXXk0
https://youtu.be/Nt-0aTfPECQ

strolll395, 2021-08-18:
I do not understand anything in this video nor does anything make sense you posted here. Go advertise somewhere else.

closed 2021-11-24 by stale[bot]