SpacehuhnTech/esp8266_deauther #1407: can't seem to able to change IP address

kdpuvvadi, 2021-02-23:
**Describe the bug**
Changed the ip address of deauther from Serial interface and rebooted the esp. tried accessing through that ip but only able to access web interface from 192.168.4.1. 

**To Reproduce**
Steps to reproduce the behavior:
1. Send "set ip 192.168.4.2" from serial monitor
2. check the settings by sending "get settings" 
3. reboot
4. Connect to WiFi of deauhter and try accessing web interface from newly set ip address
5. can't access

**Expected behavior**
Access web interface from set ip address

**Screenshots**
[image]
[image]

**Environment (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome 
 - Version 88
 - Hardware esp8266 

**Additional context**

bjornhorst, 2021-11-25:
i have the same problem

sinaj2889, 2022-05-03:
Me too opened a issue too but from web interface hope it gets fixed do you have the problem too with mac bit changing?