SpacehuhnTech/esp8266_deauther #1409: NODE MCU SSD1306 DISPLAY DIY NOT WORKING

unkn0wn-wifi, 2021-02-23:
> Please search for existing (open and closed) issues first to avoid duplicates.  
Also have a look at the [Wiki](https://github.com/spacehuhntech/esp8266_deauther/wiki).  

```
PASTE YOUR ERROR/COMPILE LOGS HERE
```
Hi there,
im reciving isses with this particular part

"If you see that the device is resetting every few seconds, check the code! Most of the time it's that you used one pin for both the display, button or LED."

I have clearly followed the wiki on how to add the ssd1306 screen and buttons but cant get an image on the screen and it rebooting. please help.

A_config.h--------

/* This software is licensed under the MIT License: https://github.com/spacehuhntech/esp8266_deauther */

#pragma once

#define ENABLE_DEBUG
#define DEBUG_PORT Serial
#define DEBUG_BAUD 115200

#define DEFAULT_ESP8266

// #define NODEMCU
// #define WEMOS_D1_MINI
// #define DISPLAY_EXAMPLE_I2C
// #define DISPLAY_EXAMPLE_SPI

// #define MALTRONICS
// #define DSTIKE_DEAUTHER_V1
// #define DSTIKE_DEAUTHER_V2
// #define DSTIKE_DEAUTHER_V3
// #define DSTIKE_DEAUTHER_V3_5
// #define DSTIKE_D_DUINO_B_V5_LED_RING
// #define DSTIKE_DEAUTHER_BOY
// #define DSTIKE_NODEMCU_07
// #define DSTIKE_NODEMCU_07_V2
// #define DSTIKE_DEAUTHER_OLED
// #define DSTIKE_DEAUTHER_OLED_V1_5_S
// #define DSTIKE_DEAUTHER_OLED_V1_5
// #define DSTIKE_DEAUTHER_OLED_V2
// #define DSTIKE_DEAUTHER_OLED_V2_5
// #define DSTIKE_DEAUTHER_OLED_V3
// #define DSTIKE_DEAUTHER_OLED_V3_5
// #define DSTIKE_DEAUTHER_OLED_V4
// #define DSTIKE_DEAUTHER_OLED_V5
// #define DSTIKE_DEAUTHER_OLED_V6
// #define DSTIKE_DEAUTHER_MOSTER
// #define DSTIKE_DEAUTHER_MOSTER_V2
// #define DSTIKE_DEAUTHER_MOSTER_V3
// #define DSTIKE_DEAUTHER_MOSTER_V4
// #define DSTIKE_DEAUTHER_MOSTER_V5
// #define DSTIKE_USB_DEAUTHER
// #define DSTIKE_USB_DEAUTHER_V2
// #define DSTIKE_DEAUTHER_WATCH
// #define DSTIKE_DEAUTHER_WATCH_V2
// #define DSTIKE_DEAUTHER_MINI
// #define DSTIKE_DEAUTHER_MINI_EVO

// #define LYASI_7W_E27_LAMP
// #define AVATAR_5W_E14_LAMP

// Forces formatting of SPIFFS and EEPROM ot startup
// #define FORMAT_SPIFFS
// #define FORMAT_EEPROM

// Forces a reset of all settings at startup
// #define RESET_SETTINGS
[... 620 lines cut ...]
 #define SPI_RES 5
 #define SPI_DC 4
 #define SPI_CS 15

   // ===== BUTTONS ===== //
 #define BUTTON_UP 255
 #define BUTTON_DOWN 255
 #define BUTTON_A 255
 #define BUTTON_B 255

   // ===== Reset ====== //
 #define RESET_BUTTON 5

   // ===== Web ===== //
 #define WEB_IP_ADDR (192, 168, 4, 1)
 #define WEB_URL "deauth.me"

 */

// ========== ERROR CHECKS ========== //
#if LED_MODE_BRIGHTNESS == 0
#error LED_MODE_BRIGHTNESS must not be zero!
#endif /* if LED_MODE_BRIGHTNESS == 0 */
[image]
[image]

Nikakz, 2021-03-16:
You uncommented this line: // #define NODEMCU

closed 2021-06-21 by stale[bot]

Ryanzinn8920, 2025-05-20:
resolveu???