SpacehuhnTech/esp8266_deauther #1413: DISPLAY_EXAMPLE_I2C.bin and endless rebooting

MegaAndrey, 2021-03-01:
Like nothing difficult, i'm downloaded bin file and installed via arduino ide, ended up getting an endless reboot of the device.

Ok, I decided to do that through the source code according to the Setup Display & Buttons (SSD1306 I2C) but the result is exactly the same, what am I doing wrong?

at the same time, the bin-file without a display (esp8266_deauther_2.6.0_NODEMCU.bin) is working properly, I can go to the web interface, attack and etc.

22:51:38.626 -> 
22:51:38.626 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,7)
22:51:38.626 -> 
22:51:38.626 -> wdt reset
22:51:38.626 -> load 0x4010f000, len 3584, room 16 
22:51:38.626 -> tail 0
22:51:38.626 -> chksum 0xb0
22:51:38.626 -> csum 0xb0
22:51:38.626 -> v2843a5ac
22:51:38.626 -> ~ld
22:51:38.719 -> 
22:51:38.719 -> Mounting SPIFFS...OK
22:51:38.719 -> Formatting SPIFFS...OK
22:51:38.812 -> Formatting EEPROM...OK
22:51:38.904 -> Loading settings...Invalid Hash
22:51:38.904 -> Settings reset to default
22:51:38.997 -> Settings saved in /settings.json
22:51:39.138 -> Device names loaded from /names.json
22:51:39.185 -> SSIDs loaded from /ssids.json
22:51:39.231 -> Scan results saved in /scan.json
22:51:39.231 -> Serial interface enabled
22:51:39.278 -> Started AP
22:51:39.278 -> [WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: false
22:51:39.278 -> STARTED! \o/
22:51:39.278 -> 2.6.0

deaut, 2021-04-27:
> Como nada difícil, descargué el archivo bin y lo instalé a través de arduino ide, terminé obteniendo un reinicio interminable del dispositivo.
> 
> Ok, decidí hacer eso a través del código fuente de acuerdo con los botones y la pantalla de configuración (SSD1306 I2C) pero el resultado es exactamente el mismo, ¿qué estoy haciendo mal?
> 
> al mismo tiempo, el archivo bin sin pantalla (esp8266_deauther_2.6.0_NODEMCU.bin) funciona correctamente, puedo ir a la interfaz web, atacar, etc.
> 
> 22:51:38.626 ->
> 22:51:38.626 -> ets Jan 8 2013,rst cause:4, boot mode:(3,7)
> 22:51:38.626 ->
> 22:51:38.626 -> wdt reset
> 22:51:38.626 -> load 0x4010f000, len 3584, room 16
> 22:51:38.626 -> tail 0
> 22:51:38.626 -> chksum 0xb0
> 22:51:38.626 -> csum 0xb0
> 22:51:38.626 -> v2843a5ac
> 22:51:38.626 -> ~ld
> 22:51:38.719 ->
> 22:51:38.719 -> Mounting SPIFFS...OK
> 22:51:38.719 -> Formatting SPIFFS...OK
> 22:51:38.812 -> Formatting EEPROM...OK
> 22:51:38.904 -> Loading settings...Invalid Hash
> 22:51:38.904 -> Settings reset to default
> 22:51:38.997 -> Settings saved in /settings.json
> 22:51:39.138 -> Device names loaded from /names.json
> 22: 51: 39.185 -> SSID cargados desde /ssids.json
> 22: 51: 39.231 -> Resultados de escaneo guardados en /scan.json
> 22: 51: 39.231 -> Interfaz serial habilitada
> 22: 51: 39.278 -> AP iniciado
> 22: 51: 39.278 -> [WiFi] Ruta: '/ web', Modo: 'AP', SSID: 'pwned', contraseña: 'deauther', canal: '1', oculto: falso, cautive-portal: falso
> 22: 51: 39.278 -> ¡COMENZADO! \ o /
> 22: 51: 39.278 -> 2.6.0

deaut, 2021-04-27:
http://192.168.4.1/ssids.html

closed 2021-08-02 by stale[bot]