SpacehuhnTech/esp8266_deauther #1424: IC2 Oled and Buttton issue

FriendlyWizard23, 2021-03-29:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

So i have an ESP8266 (the one you can purchase from AZdelivery on amazon) and an IC2 Oled Display.
After building up che circuit and uploading the code it gives me no error but there's still a problem: Buttons are not working.
The circuit is fine and i perfectly followed the scheme provided by @spacehuhn yet they do not work! The buttons's hardware is fine so i don't really know what's going on.
Another problem is that if i load the I2C Example on Deauther Config, the flash works fine but the OLED do not light up, yet the web interface works (which i suppose means the program runs fine...)

_I'll copy paste here the exact piece of the guide i'm having trouble with:

Now that your setup is done, you have to make some changes to the code.

See Installation on how to compile and flash this project using Arduino.

In Arduino under tools > Deauther Config select Display Example I2C or Display Example SPI depending on your setup

In Arduino and open the A_config.h file (second tab)

Scroll down to **#if defined(DISPLAY_EXAMPLE_I2C)** or #elif defined(DISPLAY_EXAMPLE_SPI).

If you used other pins than mentioned in the example setup above, this is where you can change them.
For example, you might need to change #define SH1106_I2C to #define SSD1306_I2C depending on the type of display you are using.
Or #define SH1106_SPI to #define SSD1306_SPI if you're using a SPI display.

In Arduino under tools > Erase Flash select All Flash Contents. This will make sure your changes are applied and override any old settings.

Now make sure you selected the correct port and upload! If something isn't working correctly, check the connections and your adjustments from step 5._

I don't really know how to solve it, right now i have the "none" deauther configuration but as i said before the buttons do not work.
And if you're wondering if it could be a GPIO problem, well i don't think so since i use the default ones.
Hopefully someone will help me out, my Telegram: @Valendrung

docteurfail, 2021-04-14:
+1 I have a similar problem no way to get the buttons working...

n0nuser, 2021-04-25:
+1 Happened the same to me on Saturday. I used the binary files and the Node2Deauther to flash it.
Switching to version 2.1.0 did work.

FriendlyWizard23, 2021-04-29:
i actually fixed it with the help of @4xtrom, if you still need explanation i can help.

jaxmayhem, 2021-05-07:
> i actually fixed it with the help of @4xtrom, if you still need explanation i can help.

I could use some help. The up and down buttons are working, but not the select button. The funny thing is, all 3 buttons are wired identically. Why are two working, and one isn't?

I'm sure it's an oversight on my part, but I haven't been able to figure it out.

Thanks in advance.

andrewbpowers, 2021-06-08:
@FriendlyWizard23 
Doesn't work!

closed 2021-09-14 by stale[bot]

beliboba, 2022-05-23:
any solution?