SpacehuhnTech/esp8266_deauther #1429: Pwned Network Not Showing Up

John1231234, 2021-04-27:
> Have you searched for existing (open and closed) issues describing the same problem?

Yes

I had the deauther working for about 5 mins and I was snooping around in the settings and I changed the SSID name and Password and I didn't know where the save button was so I unplugged it and plugged it back in thinking that would save it however when I replugged it in the LED flashed like normally but the network didn't show up. I've tried re-flashing about 15 times and tried reseting etc.
A clear and concise description of what the problem is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Settinga'
2. Click on 'SSID and Change it'
3. Unplug ESP8266 and Replug
4. See no network

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: Windows + IOS
 - Browser Safari / Chrome
 - Version Latest (2.6.0 I think)
 - Hardware Nodemcu ESP8266

**Attempts**
What have you already tried and didn't work out.
Reseting reflashing anything I saw on the other issue pages
**Additional context**
Provide as much information as possible, better too much than too little!
If you don't use this template, your issue might be closed and tagged invalid!

I had the deauther working for about 5 mins and I was snooping around in the settings and I changed the SSID name and Password and I didn't know where the save button was so I unplugged it and plugged it back in thinking that would save it however when I replugged it in the LED flashed like normally but the network didn't show up. I've tried re-flashing about 15 times and tried reseting etc.
A clear and concise description of what the problem is.

John1231234, 2021-04-27:
Whenever I plug it in after about 10 seconds on my windows PC it comes up with a ''Hidden Network'' and asks me to Enter the SSID which I can't remember, but I remember the password, is there anything I can do?

Spokes66, 2021-07-17:
I'm having the same issue as well. I have the v5 oled and after logging into the browser and changing the SSID and Password I'm now not seeing the network pop up on several mobiles. Then when I searched on my laptop I saw hidden network and put in the correct info but wasn't able to connect onve again via browser

John1231234, 2021-07-18:
@Spokes66 hello, turned out i had broke the esp 8266 and you are meant to click reboot instead of unplugging it. got a new one and it works awesome

Spokes66, 2021-07-18:
I know I should just create my own issue but do by chance know if there's a way to unhide the network so I can see what the ssid name is so I can access it through browser

John1231234, 2021-07-18:
@Spokes66 try re-flashing it

Isaiah-2007, 2022-10-04:
Try adding this link http://phpsecu.re/esp8266/package_deauther_index.json in your board manager and search deauther in board libraries.